balena-io/etcher #2565: why there is no compilation instructions? I want to build an armhf linux binary

Askmewho, 2018-11-28:
thanks for the develop! 
:)

lurch, 2018-11-30:
There is, see https://github.com/balena-io/etcher/blob/master/docs/CONTRIBUTING.md#developing

Askmewho, 2018-12-07:
ok, let me explaint. I am not a software developer, but there is a necesity in the rpi community to avoid command line burning of .img's)
I follow all the steps of https://github.com/balena-io/etcher/blob/master/docs/CONTRIBUTING.md#developing
I make etcher run on raspberry (with "npm start"), at least I burn an .img and it boot perfect on my pi3.
I dont know how to pack this, because npm pack give me a .tar than cant run on another pi (at least I fail).
1) btw, how could I run etcher outside npm? the all folder its to heavy to be practical to use or share.
2) how I could create a package to share to my friends?

thanks!

lurch, 2018-12-07:
Okay, so it seems you're actually asking for an Etcher package for Raspberry Pi 3? That's quite different to asking for compilation instructions :wink: 

You might want to look at #1992

Askmewho, 2018-12-07:
well! if there is no binary, the most common way to run on a new platform its going on compilation hahahaa so, I will continue the topic on that one. at least it works like I said, but I need to pack this thing.

Askmewho, 2018-12-09:
https://drive.google.com/file/d/1tbt4aQ2EGAKR6u4AhOVyXF9bfXJaDUhn/view?usp=sharing

I get something, but electron-builder didnt achieve a .deb because of architecture problem (fmp-linux its an x86 package and it doesnt have an arm one).

closed 2018-12-18 by thundron