balena-io/etcher #2567: Etcher can't detect by polkit daemon

ask6155, 2018-11-28:
- **Etcher version:1.4.8** 
- **Operating system and architecture: Arch Linux x86_64** 
- **Image flashed:archlinux-iso** 
- **Do you see any meaningful error information in the DevTools?** 

I tried to flash an iso and it couldn't detect my Polkit daemon.
I use i3wm so I still don't quite understand about Polkit.
I have the Polkit base package installed and the qt5 and gnome frontends installed
but it still doesn't detect any authentication agents

here is some info from Log:

```
{"stack":"Error: No polkit authentication agent found\n    at Object.exports.createError (/usr/lib/etcher/lib/shared/errors.js:253:17)\n    at Object.exports.createUserError (/usr/lib/etcher/lib/shared/errors.js:292:18)\n    at sudoPrompt.execAsync.then.catch.catch.catch.catch (/usr/lib/etcher/lib/shared/permissions.js:244:18)\n    at tryCatcher (/usr/lib/etcher/node_modules/bluebird/js/release/util.js:16:23)\n    at /usr/lib/etcher/node_modules/bluebird/js/release/catch_filter.js:34:37\n    at tryCatcher (/usr/lib/etcher/node_modules/bluebird/js/release/util.js:16:23)\n    at Promise._settlePromiseFromHandler (/usr/lib/etcher/node_modules/bluebird/js/release/promise.js:504:31)\n    at Promise._settlePromise (/usr/lib/etcher/node_modules/bluebird/js/release/promise.js:561:18)\n    at Promise._settlePromise0 (/usr/lib/etcher/node_modules/bluebird/js/release/promise.js:606:10)\n    at Promise._settlePromises (/usr/lib/etcher/node_modules/bluebird/js/release/promise.js:681:18)\n    at Async._drainQueue (/usr/lib/etcher/node_modules/bluebird/js/release/async.js:138:16)\n    at Async._drainQueues (/usr/lib/etcher/node_modules/bluebird/js/release/async.js:148:10)\n    at Async.drainQueues (/usr/lib/etcher/node_modules/bluebird/js/release/async.js:17:14)\n    at <anonymous>\n    at process._tickCallback (internal/process/next_tick.js:188:7)","message":"No polkit authentication agent found","description":"Please install a polkit authentication agent for your desktop environment of choice to continue","report":false}
```

thundron, 2018-11-28:
While we look into it, you could maybe have a look [here](https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#no-polkit-authentication-agent-found-error-in-gnulinux)

brownjohnf, 2018-11-28:
Could you also post the output of `ps -efwww | grep polkit` and `sudo systemctl status polkit`?

knuff0r, 2019-01-28:
i am using the appimage on arch linux and get the same error. running it as root works for me.

jolupa, 2019-07-06:
Still no solution for this problem. I'm in the last version running the deb package converted to CRUX package, with xfce-polkit and the problem still remains. I also run the appimage and the problem persist. All the other programs that requires polkit are working as espected.
Thanks.

bacongobbler, 2021-07-30:
Also hitting this issue. etcher must run as root before it can see the running polkit daemon.

arilebedey, 2024-03-02:
Try the following command from comment #6 in (this arch forum thread)[https://bbs.archlinux.org/viewtopic.php?id=264245]
`/usr/lib/polkit-gnome/polkit-gnome-authentication-agent-1 &`