balena-io/etcher #2568: Error message when try to open file

szotsaki, 2018-11-29:
- **Etcher version:**  {"packageType":"rpm","version":"1.4.4"}
- **Operating system and architecture:** openSUSE Tumbleweed
- **Image flashed:** --
- **Do you see any meaningful error information in the DevTools?** Just the text below

When I click on "Select image" button a popup is shown with the following text:

```
TypeError: Cannot read property 'UPDATE_NOTIFIER_SLEEP_DAYS' of undefined
    at Object.exports.notify (file:///opt/Etcher/resources/app.asar/generated/gui.js:118076:55)
    at file:///opt/Etcher/resources/app.asar/generated/gui.js:65181:27
    at tryCatcher (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/util.js:16:23)
    at Promise._settlePromiseFromHandler (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:504:31)
    at Promise._settlePromise (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:561:18)
    at Promise._settlePromise0 (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:606:10)
    at Promise._settlePromises (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:685:18)
    at Promise._fulfill (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:630:18)
    at Promise._resolveCallback (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:425:57)
    at Promise._settlePromiseFromHandler (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:516:17)
    at Promise._settlePromise (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:561:18)
    at Promise._settlePromise0 (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:606:10)
    at Promise._settlePromises (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:685:18)
    at Promise._fulfill (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:630:18)
    at MappingPromiseArray.PromiseArray._resolve (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise_array.js:125:19)
    at MappingPromiseArray._filter (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/map.js:123:10)
    at MappingPromiseArray._promiseFulfilled (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/map.js:95:18)
    at MappingPromiseArray.PromiseArray._iterate (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise_array.js:113:31)
    at MappingPromiseArray.init (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise_array.js:77:10)
    at Promise._settlePromise (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:558:21)
    at Promise._settlePromise0 (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:606:10)
    at Promise._settlePromises (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:685:18)
    at Promise._fulfill (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:630:18)
    at Promise._settlePromise (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:574:21)
    at Promise._settlePromise0 (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:606:10)
    at Promise._settlePromises (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:685:18)
    at Promise._fulfill (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:630:18)
    at Promise._resolveCallback (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:425:57)
    at Promise._settlePromiseFromHandler (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:516:17)
    at Promise._settlePromise (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:561:18)
    at Promise._settlePromise0 (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:606:10)
    at Promise._settlePromises (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:685:18)
    at Promise._fulfill (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:630:18)
    at Promise._resolveCallback (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:425:57)
    at ReductionPromiseArray._resolve (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/reduce.js:51:19)
    at Promise.completed [as _fulfillmentHandler0] (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/reduce.js:112:15)
    at Promise._settlePromise (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:558:21)
    at Promise._settlePromise0 (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:606:10)
    at Promise._settlePromises (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:685:18)
    at Async._drainQueue (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/async.js:138:16)
    at Async._drainQueues (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/async.js:148:10)
    at Async.drainQueues (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/async.js:17:14)
    at <anonymous>
    at process._tickCallback (internal/process/next_tick.js:109:7)
```

szotsaki, 2018-11-29:
It seems it's fixed in 1.4.8.

closed 2018-11-29 by szotsaki

lurch, 2018-11-30:
Duplicate of #2559