balena-io/etcher #2570: Etcher JavaScript Errors

Bean6754, 2018-11-29:
**Etcher version:** 1.4.6

**Operating system and architecture:** Windows 10 x64 and macOS x64
**Image flashed:**  N/A
**Do you see any meaningful error information in the DevTools?** Yes (see screenshots below).

## Windows.

[image]

[image]

[image]

## macOS (will add later).

### P.S: Electron DevTools is really cool!, is it this: (https://github.com/sindresorhus/electron-debug) ?

m4rky-m4rk, 2018-11-29:
Same here. It used to work. Upgrading to 1.4.8 fixed it. Windows 10 x64

Aaaagh... its now not Resin but Balena etcher. No errors, for now, but the new featured project ad annoys me as its promineltly slap in the middle of window. I prefer 1.4.6.  I had a few anxious minutes checking to see if I had an actaul offical version.

thundron, 2018-11-29:
@Bean6754 Electron includes the native DevTools you can find in Chromium by default 😉 
Closing as duplicate of https://github.com/balena-io/etcher/issues/2559

Bean6754, 2018-11-29:
Thank you. 👍 =]

Sorry for the duplicate certificate. Xx

Bean6754, 2018-11-29:
Thank you. 👍 =]

Sorry for the duplicate report. Xx

Iwfo, 2018-11-29:
Sorry my head was in the way

thundron, 2018-11-29:
@Iwfo lol no problem dude, we're taking notes and constantly refining everything so **any** feedback is *always* welcome