balena-io/etcher #2574: Cannot read property 'UPDATE_NOTIFIER_SLEEP_DAYS' of undefined     at Object.exports.notify

johnd486, 2018-12-03:
- **Etcher version:** 1.4.8 (also happened with 1.4.4)
- **Operating system and architecture:** MacOS Macbook Air 2017, Mojave 10.14.1
- **Image flashed:** None yet selected
- **Do you see any meaningful error information in the DevTools?** 

TypeError: Cannot read property 'UPDATE_NOTIFIER_SLEEP_DAYS' of undefined
    at Object.exports.notify (file:///Applications/Etcher.app/Contents/Resources/app.asar/generated/gui.js:118076:55)
    at file:///Applications/Etcher.app/Contents/Resources/app.asar/generated/gui.js:65181:27
    at tryCatcher (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/util.js:16:23)
    at Promise._settlePromiseFromHandler (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:504:31)
    at Promise._settlePromise (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:561:18)
    at Promise._settlePromise0 (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:606:10)
    at Promise._settlePromises (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:685:18)
    at Promise._fulfill (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:630:18)
    at Promise._resolveCallback (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:425:57)
    at Promise._settlePromiseFromHandler (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:516:17)
    at Promise._settlePromise (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:561:18)
    at Promise._settlePromise0 (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:606:10)
    at Promise._settlePromises (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:685:18)
    at Promise._fulfill (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:630:18)
    at MappingPromiseArray.PromiseArray._resolve (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise_array.js:125:19)

lurch, 2018-12-03:
Duplicate of #2559

thundron, 2018-12-03:
Wait wait wait, before closing, my bad:
@johnd486  did it seriously happen with 1.4.8 *too*?

johnd486, 2018-12-03:
Actually, no it doesn't ... my mistake. 1.4.8 application is called BalenaEtcher, and I was still running Etcher 1.4.4. Just trashed all copies of Etcher off my mac then reinstalled from website .... all good!

closed 2018-12-03 by johnd486

nielsbeune, 2018-12-03:
@thundron with me it seriously did happen with 1.4.8 too.
osx10.11.6 el capitan

thundron, 2018-12-03:
@nielsbeune Did you start the new BalenaEtcher app or was it still the old one?

johnd486, 2018-12-04:
> Newly downloaded. V1.4.8 for mac
> From balena etcher
>
> Niels

I re-downloaded 1.4.8 yesterday, and the fault was not there for me (MacOS Mojave 10.14.1). The app name change caught me out, so I mis-reported a fault with 1.4.8 when in fact I was still running 1.4.4.

John

lurch, 2018-12-04:
IIRC the Etcher version gets displayed near the start of the DevToolsConsole log?