balena-io/etcher #2575: Installing 1.4.8 keeps previous version

kayone, 2018-12-05:
- **Etcher version: 1.4.8** 
- **Operating system and architecture: Windows 10 1803 x86** 

Hi Guys, 

I'm the maintainer for https://github.com/appget/appget/; we got a report of etcher not upgrading correctly. After doing some investigation, I realized the new release of etcher (I'm guessing since the name change) doesn't remove previous versions of the app. 

This can cause confusion for users (as it did in this case) since the user could just keep running the old version of etcher while being under the impression that they are running the new version they just installed.

related: https://github.com/appget/appget/issues/16

lurch, 2018-12-05:
See also #2554

thundron, 2018-12-17:
@kayone You're right about the (company) name change, although electron-builder doesn't let us customize the packages in order to remove other applications (yet) so the best thing is just to link the new executable. I'll update the documentation so that it states the correct executable `BalenaEtcher` but that's as far as we can go for now unfortunately with the current lack of automated options on this side

closed 2018-12-28 by thundron