balena-io/etcher #2590: Windows CLI not working at all

kolegs, 2018-12-30:
- **Etcher version:** Windows x64 1.9.4
- **Operating system and architecture:** Windows 10 Home 1809
- **Do you see any meaningful error information in the DevTools?** 
`c:\snapshot>balena-etcher.exe f:\Pobrane\ubuntu-18.04.1-desktop-amd64.iso                                               

pkg/prelude/bootstrap.js:1080                                                                                                 
throw error;                                                                                                            ^                                                                                                                                                                                                                                         Error: Could not locate the bindings file. Tried:                                                                        
→ C:\snapshot\balena-etcher-cli-1.4.9-windows-x64-app\node_modules\mountutils\build\MountUtils.node                     
→ C:\snapshot\balena-etcher-cli-1.4.9-windows-x64-app\node_modules\mountutils\build\Debug\MountUtils.node               
→ C:\snapshot\balena-etcher-cli-1.4.9-windows-x64-app\node_modules\mountutils\build\Release\MountUtils.node             
→ C:\snapshot\balena-etcher-cli-1.4.9-windows-x64-app\node_modules\mountutils\out\Debug\MountUtils.node                 
→ C:\snapshot\balena-etcher-cli-1.4.9-windows-x64-app\node_modules\mountutils\Debug\MountUtils.node                     
→ C:\snapshot\balena-etcher-cli-1.4.9-windows-x64-app\node_modules\mountutils\out\Release\MountUtils.node               
→ C:\snapshot\balena-etcher-cli-1.4.9-windows-x64-app\node_modules\mountutils\Release\MountUtils.node                   
→ C:\snapshot\balena-etcher-cli-1.4.9-windows-x64-app\node_modules\mountutils\build\default\MountUtils.node             
→ C:\snapshot\balena-etcher-cli-1.4.9-windows-x64-app\node_modules\mountutils\compiled\6.10.3\win32\x64\MountUtils.node                                                                                                                            at bindings (C:\snapshot\balena-etcher-cli-1.4.9-windows-x64-app\node_modules\bindings\bindings.js:93:9)                
at Object.<anonymous> (C:\snapshot\balena-etcher-cli-1.4.9-windows-x64-app\node_modules\mountutils\index.js:52:37)      
at Module._compile (pkg/prelude/bootstrap.js:1151:22)                                                                   
at Object.Module._extensions..js (module.js:579:10)                                                                     
at Module.load (module.js:487:32)                                                                                       
at tryModuleLoad (module.js:446:12)                                                                                     
at Function.Module._load (module.js:438:3)                                                                              
at Module.require (module.js:497:17)                                                                                    
at Module.require (pkg/prelude/bootstrap.js:1061:31)                                                                    
at require (internal/module.js:20:19)                                                                                   
at Object.<anonymous> (C:\snapshot\balena-etcher-cli-1.4.9-windows-x64-app\lib\sdk\writer\index.js:0)                   at Module._compile (pkg/prelude/bootstrap.js:1151:22)                                                                   
at Object.Module._extensions..js (module.js:579:10)                                                                     
at Module.load (module.js:487:32)                                                                                       
at tryModuleLoad (module.js:446:12)                                                                                     
at Function.Module._load (module.js:438:3)                                                                              
at Module.require (module.js:497:17)                                                                                    
at Module.require (pkg/prelude/bootstrap.js:1061:31)                                                                    
at require (internal/module.js:20:19)                                                                                   
at Object.<anonymous> (C:\snapshot\balena-etcher-cli-1.4.9-windows-x64-app\lib\cli\etcher.js:0)                         at Module._compile (pkg/prelude/bootstrap.js:1151:22)                                                                   
at Object.Module._extensions..js (module.js:579:10)                                                                     
at Module.load (module.js:487:32)                                                                                       
at tryModuleLoad (module.js:446:12)                                                                                     
at Function.Module._load (module.js:438:3)                                                                              
at Module.runMain (pkg/prelude/bootstrap.js:1214:12)                                                                    
at run (bootstrap_node.js:432:7)                                                                                        
at startup (bootstrap_node.js:192:9)                                                                                    
at bootstrap_node.js:547:3 `

What is more I moved the file so it is in the path it looks for it.
`c:\snapshot>dir C:\snapshot\balena-etcher-cli-1.4.9-windows-x64-app\node_modules\mountutils\build\Release\MountUtils.node                                                                                                                                                                                                                                                                                                                                    Directory of C:\snapshot\balena-etcher-cli-1.4.9-windows-x64-app\node_modules\mountutils\build\Release                                                                                                                                         28/12/2018  17:43           302,592 MountUtils.node                                                                                    1 File(s)        302,592 bytes                                                                                          0 Dir(s)  21,620,035,584 bytes free                                                                                                                 `

lurch, 2018-12-31:
Duplicate of #2274 ?

thundron, 2018-12-31:
Apparently so, we're still investigating it as it looks like a regression in the CI which was reworked recently. Closing this one, refer to the other for updates

closed 2018-12-31 by thundron