balena-io/etcher #2596: lzma_native.node: undefined symbol

adonespitogo, 2019-01-08:
- Etcher version: etcher 1.4.8-1
- Electron version: electron 3.0.13-2
- Operating system and architecture: Arch Linux x64
- Devtools error:

```javascript
Uncaught Error: /usr/lib/etcher/node_modules/lzma-native/binding/lzma_native.node: undefined symbol: _ZNK2v88Function11NewInstanceEiPNS_5LocalINS_5ValueEEE
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:166:20)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:740)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:166:20)
    at Module.load (internal/modules/cjs/loader.js:620)
    at tryModuleLoad (internal/modules/cjs/loader.js:559)
    at Function.Module._load (internal/modules/cjs/loader.js:551)
    at Module.require (internal/modules/cjs/loader.js:658)
    at require (internal/modules/cjs/helpers.js:20)
    at /usr/lib/etcher/node_modules/lzma-native/index.js:11
    at Object.<anonymous> (/usr/lib/etcher/node_modules/lzma-native/index.js:623)
    at Object.<anonymous> (/usr/lib/etcher/node_modules/lzma-native/index.js:625)
    at Module._compile (internal/modules/cjs/loader.js:711)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:722)
    at Module.load (internal/modules/cjs/loader.js:620)
    at tryModuleLoad (internal/modules/cjs/loader.js:559)
    at Function.Module._load (internal/modules/cjs/loader.js:551)
/usr/lib/electron/resources/electron.asar/renderer/security-warnings.js:184 Electron Security Warning (Insecure Content-Security-Policy) This renderer process has either no Content Security
      Policy set or a policy with "unsafe-eval" enabled. This exposes users of
      this app to unnecessary security risks.
 
For more information and help, consult
https://electronjs.org/docs/tutorial/security.
 This warning will not show up
once the app is packaged.
warnAboutInsecureCSP @ /usr/lib/electron/resources/electron.asar/renderer/security-warnings.js:184
VM530:1 Uncaught TypeError: require(...).webFrame.setZoomLevelLimits is not a function
    at <anonymous>:1:30
    at EventEmitter.electron.ipcRenderer.on (/usr/lib/electron/resources/electron.asar/renderer/web-frame-init.js:36)
    at EventEmitter.emit (events.js:182)
```

I'm having this issue right now after fresh install, although not totally blank screen. Only top menus show up:

https://imgur.com/MMiYIVj

https://imgur.com/TtP4uU0

thundron, 2019-01-09:
@adonespitogo We don't support electron 3 yet but iirc that's what the arch etcher package uses, so you might want to download another one instead

thundron, 2019-01-09:
You can follow progress [here](https://github.com/balena-io/etcher/pull/2578). Closing, but please comment again if you have the same issue using another package

closed 2019-01-09 by thundron