balena-io/etcher #2597: Incomplete list of target drives on OSX

pskowronek, 2019-01-08:
- balenaEtcher v. 1.4.9
- macOS 10.14.2
- image - irrelevant but it was Raspbian Stretch with desktop and recommended software (~4GB)

I've tried to use balenaEtcher on OSX and the application is unable to list SD Card that is put into built-in SD Card reader on MacBook Pro (13-inch, Mid 2012). The mount command in terminal shows this card as follows: `/dev/disk6s1 on /Volumes/SAMSUNG32 (msdos, local, nodev, nosuid, noowners)`.

What is interesting that Etcher v1.2.1 list this drive w/o any problems. The other problem with the newest version is that is started to show OSX "sparse bundle drives".

[image]
[image]

lurch, 2019-01-08:
Sparse bundles were also mentioned in #2379 

(FYI https://github.com/balena-io-modules/drivelist is what Etcher uses to list drives)

thundron, 2019-01-23:
Could you please paste here the output of `diskutil list -plist` after inserting your sdcard with Etcher running?

pskowronek, 2019-01-28:
In the attached file you can find "sanitized" output from the command.

The problem though is that now I can see that SD card that is put into the card reader - this is exactly the same SD card but it was reformatted (using older Etcher, in the attached logs and screen shot it is now named: 'boot'). So, there are two possibilities:
1) SD card was strangely formatted by manufacturer (Samsung 32 EVO plus) - for sure it was FAT, but maybe it had some additional partitions?
2) macOS update 10.14.3.

The problem with Sparse Bundle Drives is still there, what's more on the list I see also DMG file that is being mounted (not sure if by design):

[image]

[output from diskutil command](https://github.com/balena-io/etcher/files/2804294/output.txt)

I will try to get a brand new Samsung 32 EVO plus - stay tuned :)

thundron, 2021-02-02:
We have improved the UI and fixed some unexpected behaviours in more recent versions, please try those. Closing as this version is very outdated

closed 2021-02-02 by thundron

pskowronek, 2021-02-06:
FYI, verified on 1.5.116 - now works as expected.