balena-io/etcher #2598: SD card shows image, SD card is empty

GabrielBalan90, 2019-01-09:
- **Etcher version:** 
- **Operating system and architecture:** 
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?**

thundron, 2019-01-09:
Probably a duplicate of https://github.com/balena-io/etcher/issues/2576
Could you please paste here the output of `lsblk --all --json --paths --output-all` ?

lurch, 2019-01-09:
@thundron `drivelist` actually runs `lsblk --bytes --all --json --paths --output-all` :wink: 
@GabrielBalan90 If that doesn't work, paste the output of `lsblk --bytes --pairs --all`

GabrielBalan90, 2019-01-09:
@thundron I've used your output:

gabriel@debian:~$ lsblk --bytes --pairs --all
NAME="sda" MAJ:MIN="8:0" RM="0" SIZE="500107862016" RO="0" TYPE="disk" MOUNTPOINT=""
NAME="sda2" MAJ:MIN="8:2" RM="0" SIZE="250000441344" RO="0" TYPE="part" MOUNTPOINT="/mnt/sda2"
NAME="sda3" MAJ:MIN="8:3" RM="0" SIZE="250000441344" RO="0" TYPE="part" MOUNTPOINT="/mnt/sda3"
NAME="sdb" MAJ:MIN="8:16" RM="0" SIZE="80026361856" RO="0" TYPE="disk" MOUNTPOINT=""
NAME="sdb1" MAJ:MIN="8:17" RM="0" SIZE="29998710784" RO="0" TYPE="part" MOUNTPOINT="/"
NAME="sdb2" MAJ:MIN="8:18" RM="0" SIZE="45999980544" RO="0" TYPE="part" MOUNTPOINT="/home"
NAME="sdb3" MAJ:MIN="8:19" RM="0" SIZE="4026531840" RO="0" TYPE="part" MOUNTPOINT="[SWAP]"
NAME="sdc" MAJ:MIN="8:32" RM="1" SIZE="32010928128" RO="0" TYPE="disk" MOUNTPOINT=""
NAME="sdc1" MAJ:MIN="8:33" RM="1" SIZE="32006733824" RO="0" TYPE="part" MOUNTPOINT="/media/gabriel/294D-C7BB"
NAME="sr0" MAJ:MIN="11:0" RM="1" SIZE="1073741312" RO="0" TYPE="rom" MOUNTPOINT=""
gabriel@debian:~$

lurch, 2019-01-10:
Looks like this is indeed the `drivelist` bug which I identified in my "EDIT2" here: https://github.com/balena-io/etcher/issues/2576#issuecomment-449993159

Because `/dev/sda` and `/dev/sdb` and `/dev/sdc` all have the same MAJor number of `8`, drivelist erroneously thinks they're all the same drive :disappointed: :floppy_disk:

thundron, 2019-01-10:
Yes they all have the same mountpoint when I force to use the other route on my Linux VM :/

thundron, 2019-01-10:
I'm working on it btw, just so you know. Closing this, refer to the linked issue above

closed 2019-01-10 by thundron