balena-io/etcher #2599: Option in settings to remove Ads

dhoney, 2019-01-09:
- **Etcher version:** 1.4.9
- **Operating system and architecture:** Mac OSX
- **Image flashed:** fincm3
- **Do you see any meaningful error information in the DevTools?** No

I would like the team to consider adding a feature in the settings panel to choose not to show the Ads that show up in Balena-Etcher when flashing an image. 
I find the functionality distracting and taking up unnecessary screen real estate when I'm monitoring my image flash progress.

Thanks!

lurch, 2019-01-10:
https://github.com/balena-io/etcher/pull/2553#issuecomment-451862679

DeviousPenguin, 2019-01-17:
I didn't want this feature either, I just went back to 1.4.7 ;)

starquake, 2019-01-19:
I'm still contemplating if getting annoyed by this is fair or not.

But to me it feels like my favorite no-nonsense image writing tool just gained nonsense...

alexandrosm, 2019-01-21:
Hey there, balena founder here. The balena team (formerly known as resin.io) started and to this day develops etcher, paying full time developers, servers, bandwidth, etc, while fully open sourcing the project and offering it for free to everyone.

We're trying to find a model that justifies us continuing to spend hundreds of thousands of dollars a year on development of etcher while keeping it free and open source. We feel that the featured project is interesting to the maker audience (this isn't exactly a McDonald's ad) while helping us at balena promote our other products which are also very interesting to the same folks that use etcher. As a data point, many many people read the project page every day, and many hundreds have signed up and are building projects with balena cloud. This confirms for us that the featured project indeed useful and interesting.

The next steps we want to make is to feature third party projects, not just ones made by the balena team.

We also want to continue to iterate on the UI to make it blend better, we know there's usability concerns, this is still an early phase of the effort. For the next iteration we want to make sure the UX is better, especially for people who are writing up tutorials

starquake, 2019-01-22:
Let me first state: I am a fan of etcher and would like it to stay that way. And I understand why you are looking for ways to recoup the costs. I'm not attacking balena but the decision to add promotions is going to stir up questions. So I might as well be the one to start the discussion.

Have you considered patreon or donations? Have you considered constraining your promotions to the download page, README, about dialog and so on?

IMHO the addition of promotions just opens a can of worms:
- Who is the arbiter of which projects get promoted?
- Is it allowed to add promotions to projects that do not use any of your products?
- Can contributors that are not part of your company get promotion of their projects in exchange for their contributions?
- Is it allowed to fork the project and remove the promotions code?

I hope you can find a way to both the recoup the costs and keep all user, makers and contributors happy.

alexandrosm, 2019-01-22:
The arbiter for these projects will be balena, the projects will generally use balena products, and putting them somewhere nobody will see them kinda defeats the point :)

The project is open source Apache 2.0. You can do anything the license permits, including forking.

EpicLPer, 2019-01-22:
Make a Patreon, done... Ads are just stupid and will drive people to make a ad-free fork of this...

dglatting, 2019-01-23:
I solved the ad annoyance problem by deleting Etcher from my system. I have other choices.

kriekprinsloo, 2019-01-23:
Hello,
I've been a user of Etcher for a long, long time.  I think it's fair for a company who's maintaining and improving an open-source tool for burning SD cards (and other media devices) to try and promote some work, trying to keep their company up in the air.  I know the opinion of a fifteen-year-old isn't usually taken seriously on the internet, but just think;  it's a small price to pay for a great tool, I certainly don't mind it, and it could be worse.

Thank you for your time.

andybarnes, 2019-01-24:
A lot of what made etcher so good was the clean lines and no cruft. Guess I'm looking for a new tool and I'll just have to roll back to 1.4.7  in the meantime.

In a related note renaming the binary from etcher-electron to balena-etcher-electron also broke a ton of my scripts and I'll likely replace rather than fix those up, very inconsiderate!

scottie-uk, 2019-01-25:
If someone does decide to fork it maybe they can do something about the large amount of outgoing connections that etcher seems to have. I get it however, Ballenna want Etcher to pay its keep via advertising and analytics. However, not on my computer they wont.

angela-d, 2019-02-07:
They're exorbitantly loud and it feels like I'm being yelled at.  I had to minimize my window at work today, rather than watch the progress on my 2nd monitor like I normally do.  I couldn't even tell you what the ad was about.. something about a Raspberry Pi?  The *text* is the only thing that caught my attention.  Why can't these be inconspicuous textual references to interesting projects??

These types of ads are **not** effective, unless your audience is new to the web.

After spending my teen years and adulthood on the web, I've trained myself to **ignore** garbage like this, as have most people who probably won't waste their time complaining, they'll just look for another passion project that doesn't bombard their users with junk ads.

It's also curious why a project like this would also collect analytics.  What data points are you collecting and why?  For projects that have an **opt-in** box that *explains* what they do with the data, I almost always oblige.  Those that silently do it end up on my hosts filter and I also go out of my way to submit the offenders to curated filter lists.

Kurgol, 2019-02-07:
rolled back to 1.4.7 while I find a replacement, disapointing.

rosshiga, 2019-03-05:
Would a small crypto miner work and pay out? I think it's fair for compensation. Developers have a right to seek value for their work. Most people here are concerned about the privacy considerations. We do not want cookies or tracking or the like. If you clearly state a private policy and protect our privacy I fully support these changes.

nyanshak, 2019-05-31:
I don't think this fixes all of the advertising, but it's a start:

In `${XDG_CONFIG_HOME}/balena-etcher-electron/config.json`, add this line:
```featuredProjectEndpoint": "http://127.0.0.1"```

rradar, 2019-06-19:
I really hope there will be a privacy friendly (actually to **comply with** eu **laws** would already be a start #2766) and ad free fork which concentrates on burning and verifying images - nothing else.

etcher is clearly going into the wrong directions recently! :-1:

thundron, 2019-06-20:
@rradar We're working on it, both privacy-friendly settings and EU laws compliance 👍 it just requires more time and internal discussions before jumping into implementation (and activity-tracking) as opposed to what we had already planned for other functionalities in the roadmap.

TL;DR: some more patience

rradar, 2019-07-13:
@thundron General Data Protection Regulation (GDPR) were implemented mandatory over a year ago in may 2018. The following sanctions (https://en.wikipedia.org/wiki/General_Data_Protection_Regulation#Sanctions) can be enforced. Maybe we soon find balena on this list: https://en.wikipedia.org/wiki/GDPR_fines_and_notices (like the cool kids from google!) 

TL;DR: How long does a balena inc needs to comply with (mandatory) laws?

saitam1, 2019-07-19:
http://www.enforcementtracker.com/

rradar, 2019-08-12:
Greece (who is in charge for balena inc based in athen) just catched their first fish :fish: and collected **one hundred and fifty thousand Euros** (EUR 150,000.00) from a company violating the GDPR (like balena :whale:)

Looks like they are awake now! Hopefully nobody writes them a mail (complaints@dpa.gr) and tells about this violation here :zipper_mouth_face:

rradar, 2019-08-25:
https://github.com/standard/standard/issues/1381

EpicLPer, 2019-08-25:
I could also hint at them in Austria if necessary ;)

petrosagg, 2019-09-14:
Since the original question about removing the feature projects was addressed and the discussion moved to data collection concerns I'm closing this in favour of #2766

closed 2019-09-14 by petrosagg

nyanshak, 2019-09-14:
Opened #2903 as I don't believe this has been resolved.

Kurgol, 2019-09-16:
blatantly obvious they don't care, have abandoned in favor of other projects. Such a shame as it used to be excellent.

petrosagg, 2019-09-16:
@Kurgol we care about the users of Etcher as a whole. I have explained the reasoning of the decision for this in #2903, but I will elaborate a bit more.

I assume you would agree that an abandoned project would be a bad outcome for Etcher users. Doing that would be not caring about our users. The fact of the matter is that 98.5% of all commits ever made to Etcher is from balena employees. You can't just ignore this fact.

This is the reason that it's excellent and will continue to be. I understand that you may not like being shown a tutorial for a project while flashing a card, but you have to acknowledge that the continued investment in the development of Etcher is directly benefiting the users.

Removing the featured project would be not caring about Etcher having a sustained future. If there is an argument why the featured project making more damage than good for the average user we'd be very interested to hear. Acting on the will of a single user while ignoring everyone else is akin to doing statistical claims using anecdotes.

On GDPR, please see the detailed comment here https://github.com/balena-io/etcher/issues/2766#issuecomment-531437542.

rradar, 2019-09-20:
Where is **the** ad free etcher fork? :eyes:

lurch, 2019-09-20:
@rradar There's nothing stopping you making one, if you feel so strongly...

rradar, 2019-09-29:
@lurch sadly I'm don't have skills for that... but for now I'm using privacy respecting software which don't transmit data on their own (and without permission!) when granted internet access...

petrosagg, 2019-09-29:
@rradar we don't consider the etcher featured project to be ads and we have no intention of removing them for reasons explained above.

Incidentally, this happened yesterday https://www.reddit.com/r/raspberry_pi/comments/dahfkq/balenasound_project_working/ . In case you wanted concrete evidence that these "ads" are not evil and benefit the community.

In any case, you're free to use whatever software you want. We'll be here if you change your mind :)

Locking the thread as it has gone offtopic