balena-io/etcher #2602: Wrong size when selecting a volume

matiasdecarli, 2019-01-10:
- 1.4.9
- macOS Mojave 10.14.2 
- 2018-11-13-raspbian-stretch.img

Even when my SD card is 32GB

[image]

I'm seeing the size on Etcher as 2TB

[image]

thundron, 2019-01-16:
We're having a look, but that looks weird. I know this might sound very superficial, but are you *sure* that it's the same drive you're talking about? Could you please paste here the output of `lsblk --bytes --all --json --paths --output-all` after you connect your unit?

matiasdecarli, 2019-01-22:
Yes, I'm sure since I didn't have any other thing connected on the USB or on the memory card. Either way, I could flash the SD card and it's working OK

The output of the command (even after the flashed is showing 2.2TB) is this

```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>AllDisks</key>
	<array>
		<string>disk0</string>
		<string>disk0s1</string>
		<string>disk0s2</string>
		<string>disk1</string>
		<string>disk1s1</string>
		<string>disk1s2</string>
		<string>disk1s3</string>
		<string>disk1s4</string>
		<string>disk2</string>
		<string>disk2s1</string>
		<string>disk2s2</string>
	</array>
	<key>AllDisksAndPartitions</key>
	<array>
		<dict>
			<key>Content</key>
			<string>GUID_partition_scheme</string>
			<key>DeviceIdentifier</key>
			<string>disk0</string>
			<key>Partitions</key>
			<array>
				<dict>
					<key>Content</key>
					<string>EFI</string>
					<key>DeviceIdentifier</key>
					<string>disk0s1</string>
					<key>DiskUUID</key>
					<string>391B2326-A49D-45E9-BC2F-3902D54B4EAA</string>
					<key>Size</key>
					<integer>314572800</integer>
					<key>VolumeName</key>
					<string>EFI</string>
					<key>VolumeUUID</key>
					<string>E783267B-A4C3-3556-B751-DBED770EB996</string>
				</dict>
				<dict>
					<key>Content</key>
					<string>Apple_APFS</string>
					<key>DeviceIdentifier</key>
					<string>disk0s2</string>
					<key>DiskUUID</key>
					<string>ADE6300D-3F94-4D72-A744-B59B7FD1BB06</string>
					<key>Size</key>
					<integer>499963174912</integer>
				</dict>
			</array>
			<key>Size</key>
			<integer>500277792768</integer>
		</dict>
		<dict>
			<key>APFSVolumes</key>
			<array>
				<dict>
					<key>DeviceIdentifier</key>
					<string>disk1s1</string>
					<key>DiskUUID</key>
					<string>A7D9D66B-6BBA-3948-9704-020E0E514F67</string>
					<key>MountPoint</key>
					<string>/</string>
					<key>Size</key>
					<integer>499963174912</integer>
					<key>VolumeName</key>
					<string>Macintosh HD</string>
					<key>VolumeUUID</key>
					<string>A7D9D66B-6BBA-3948-9704-020E0E514F67</string>
				</dict>
				<dict>
					<key>DeviceIdentifier</key>
					<string>disk1s2</string>
					<key>DiskUUID</key>
					<string>3EC765EE-02D9-4A2B-9BC8-43D65F575AFA</string>
					<key>Size</key>
					<integer>499963174912</integer>
					<key>VolumeName</key>
					<string>Preboot</string>
					<key>VolumeUUID</key>
					<string>3EC765EE-02D9-4A2B-9BC8-43D65F575AFA</string>
				</dict>
				<dict>
					<key>DeviceIdentifier</key>
					<string>disk1s3</string>
					<key>DiskUUID</key>
					<string>F62FE2F8-1631-49D0-B3E8-277F2D992406</string>
					<key>MountPoint</key>
					<string>/Volumes/Recovery</string>
					<key>Size</key>
					<integer>499963174912</integer>
					<key>VolumeName</key>
					<string>Recovery</string>
					<key>VolumeUUID</key>
					<string>F62FE2F8-1631-49D0-B3E8-277F2D992406</string>
				</dict>
				<dict>
					<key>DeviceIdentifier</key>
					<string>disk1s4</string>
					<key>DiskUUID</key>
					<string>74C4D718-5053-4A7D-8AA0-1872C93CD751</string>
					<key>MountPoint</key>
					<string>/private/var/vm</string>
					<key>Size</key>
					<integer>499963174912</integer>
					<key>VolumeName</key>
					<string>VM</string>
					<key>VolumeUUID</key>
					<string>74C4D718-5053-4A7D-8AA0-1872C93CD751</string>
				</dict>
			</array>
			<key>Content</key>
			<string>EF57347C-0000-11AA-AA11-00306543ECAC</string>
			<key>DeviceIdentifier</key>
			<string>disk1</string>
			<key>Partitions</key>
			<array/>
			<key>Size</key>
			<integer>499963174912</integer>
		</dict>
		<dict>
			<key>Content</key>
			<string>FDisk_partition_scheme</string>
			<key>DeviceIdentifier</key>
			<string>disk2</string>
			<key>Partitions</key>
			<array>
				<dict>
					<key>Content</key>
					<string>Windows_FAT_32</string>
					<key>DeviceIdentifier</key>
					<string>disk2s1</string>
					<key>MountPoint</key>
					<string>/Volumes/boot</string>
					<key>Size</key>
					<integer>45190144</integer>
					<key>VolumeName</key>
					<string>boot</string>
					<key>VolumeUUID</key>
					<string>9127C786-0F71-3A3A-9ED4-345111F7B6B6</string>
				</dict>
				<dict>
					<key>Content</key>
					<string>Linux</string>
					<key>DeviceIdentifier</key>
					<string>disk2s2</string>
					<key>Size</key>
					<integer>31976325120</integer>
				</dict>
			</array>
			<key>Size</key>
			<integer>2196690173952</integer>
		</dict>
	</array>
	<key>VolumesFromDisks</key>
	<array>
		<string>Macintosh HD</string>
		<string>Recovery</string>
		<string>VM</string>
		<string>boot</string>
	</array>
	<key>WholeDisks</key>
	<array>
		<string>disk0</string>
		<string>disk1</string>
		<string>disk2</string>
	</array>
</dict>
</plist>
```

This is the model of the Memory Card:

[image]

Let me know if I can provide you with more information

Thanks!

lurch, 2019-01-22:
```
			<key>DeviceIdentifier</key>
			<string>disk2</string>
[snip]
			<key>Size</key>
			<integer>2196690173952</integer>
```
2196690173952 bytes is indeed 2.2TB - how weird! :confused: Maybe an OSX bug?

thundron, 2019-01-23:
Definitely something to look into OSX rather than Etcher. The closest thing I could find is [this guide](https://www.macworld.com/article/3243205/storage/how-to-resize-disk-partition-in-macos.html) to resize partitions but other than this, I have no clue.
Closing as unrelated to us, but feel free to update the issue with more infos if you're willing to!

closed 2019-01-23 by thundron