balena-io/etcher #2603: Exception dialog on start

paul-hammant, 2019-01-11:
- **Etcher version:** 1.4.4
- **Operating system and architecture:** Mac: 10.13.6
- **Image flashed:** n/a
- **Do you see any meaningful error information in the DevTools?** 

[image]

yup, here's the snip:

```
 _____ _       _
|  ___| |     | |
| |__ | |_ ___| |__   ___ _ __
|  __|| __/ __| '_ \ / _ \ '__|
| |___| || (__| | | |  __/ |
\____/ \__\___|_| |_|\___|_|

Interested in joining the Etcher team?
Drop us a line at join+etcher@resin.io

Version = 1.4.4, Type = dmg
gui.js:105081 Fri Jan 11 2019 23:31:40 GMT+0000 (GMT) Application start ({"packageType":"dmg","version":"1.4.4"})
gui.js:105081 Fri Jan 11 2019 23:31:40 GMT+0000 (GMT) Checking for updates ({"currentVersion":"1.4.4","stable":true,"updateSemverRange":"<2.0.0","includeUnstableChannel":false})
gui.js:105081 etcher:sdk:scanner start +142ms true
gui.js:105081 Fri Jan 11 2019 23:31:43 GMT+0000 (GMT) Notifying update ({"latestVersion":"1.4.8"})
gui.js:105081 {"stack":"TypeError: Cannot read property 'UPDATE_NOTIFIER_SLEEP_DAYS' of undefined\n    at Object.exports.notify (file:///Applications/Etcher.app/Contents/Resources/app.asar/generated/gui.js:118076:55)\n    at file:///Applications/Etcher.app/Contents/Resources/app.asar/generated/gui.js:65181:27\n    at tryCatcher (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/util.js:16:23)\n    at Promise._settlePromiseFromHandler (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:504:31)\n    at Promise._settlePromise (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:561:18)\n    at Promise._settlePromise0 (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:606:10)\n    at Promise._settlePromises (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:685:18)\n    at Promise._fulfill (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:630:18)\n    at Promise._resolveCallback (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:425:57)\n    at Promise._settlePromiseFromHandler (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:516:17)\n    at Promise._settlePromise (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:561:18)\n    at Promise._settlePromise0 (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:606:10)\n    at Promise._settlePromises (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:685:18)\n    at Promise._fulfill (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:630:18)\n    at MappingPromiseArray.PromiseArray._resolve (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise_array.js:125:19)\n    at MappingPromiseArray._filter (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/map.js:123:10)\n    at MappingPromiseArray._promiseFulfilled (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/map.js:95:18)\n    at MappingPromiseArray.PromiseArray._iterate (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise_array.js:113:31)\n    at MappingPromiseArray.init (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise_array.js:77:10)\n    at Promise._settlePromise (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:558:21)\n    at Promise._settlePromise0 (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:606:10)\n    at Promise._settlePromises (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:685:18)\n    at Promise._fulfill (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:630:18)\n    at Promise._settlePromise (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:574:21)\n    at Promise._settlePromise0 (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:606:10)\n    at Promise._settlePromises (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:685:18)\n    at Promise._fulfill (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:630:18)\n    at Promise._resolveCallback (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:425:57)\n    at Promise._settlePromiseFromHandler (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:516:17)\n    at Promise._settlePromise (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:561:18)\n    at Promise._settlePromise0 (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:606:10)\n    at Promise._settlePromises (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:685:18)\n    at Promise._fulfill (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:630:18)\n    at Promise._resolveCallback (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:425:57)\n    at ReductionPromiseArray._resolve (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/reduce.js:51:19)\n    at Promise.completed [as _fulfillmentHandler0] (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/reduce.js:112:15)\n    at Promise._settlePromise (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:558:21)\n    at Promise._settlePromise0 (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:606:10)\n    at Promise._settlePromises (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:685:18)\n    at Async._drainQueue (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/async.js:138:16)\n    at Async._drainQueues (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/async.js:148:10)\n    at Async.drainQueues (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/async.js:17:14)\n    at <anonymous>\n    at process._tickCallback (internal/process/next_tick.js:109:7)","message":"Cannot read property 'UPDATE_NOTIFIER_SLEEP_DAYS' of undefined"}
console.(anonymous function) @ gui.js:105081
gui.js:105081 Fri Jan 11 2019 23:31:57 GMT+0000 (GMT) [object Event]
gui.js:105081 Fri Jan 11 2019 23:31:57 GMT+0000 (GMT) <link rel=preload> must have a valid `as` value
```

lurch, 2019-01-12:
I believe that 1.4.4 is now an old version, and this particular problem is fixed in newer Etchers?

paul-hammant, 2019-01-12:
Yup seems fixed in 1.4.9.

closed 2019-01-12 by paul-hammant