balena-io/etcher #2604: [Feature Request] Convert from raw .img format to bootable .iso

pcgeek86, 2019-01-15:
## Summary

As a balenaCloud user, I would like an easier mechanism to install balenaOS onto virtual machines (ie. Hyper-V, VMware, VirtualBox). If I could convert the balenaOS "flasher" image from raw .img format to a bootable ISO file, this would give me more flexibility in installing balenaOS on virtual machines, and minimize the toolchain and configuration required to get a VM up and running.

## Current State

Today, I have to follow a complex process to convert the balenaOS "flasher" image to VHDX/VMDK disk formats, mount this secondary disk as a bootable device, boot the system, and then later remove the "flasher" disk, once the OS has been "staged" onto a secondary disk of a virtual machine. This process is cumbersome, and requires conversion between disk formats (ie. VHDX, VMware Workstation-compatible VMDK, VMware ESX-compatible VMDK, etc.). Because there are two separate disks attached to the VM, I also have to worry about the disk boot order, to ensure that the virtual machine doesn't try to repeatedly boot from the "flasher" disk, if it's still attached after the staging process.

## Desired State

Instead, I'd like the option to convert the .img directly to a bootable .iso, which is more easily connected / disconnected from a virtual machine, and doesn't require conversion between a variety of disk formats. It would be great if Etcher could perform this .img to .iso conversion for me.

---

**NOTE**: Perhaps it makes more sense for this to be a feature request against balenaCloud, to provide a bootable ISO download option, but I figured I would post it here to start the discussion. It might actually be more useful to provide a generic tool (Etcher) to perform these conversions in scenarios outside of balenaCloud as well.

lurch, 2019-01-15:
> Perhaps it makes more sense for this to be a feature request against balenaCloud

I no longer work on Etcher, but IMHO https://github.com/balena-os/balena-os/issues is the best place for this request.

pcgeek86, 2019-01-15:
>> Perhaps it makes more sense for this to be a feature request against balenaCloud

> I no longer work on Etcher, but IMHO https://github.com/balena-os/balena-os/issues is the best place for this request. 

Even with this in mind?

> It might actually be more useful to provide a generic tool (Etcher) to perform these conversions in scenarios outside of balenaCloud as well.

thundron, 2019-01-15:
@pcgeek86 I have to agree with @lurch for now: we are considering bringing something *similar* but quite different at the same time which you can get more info about [here](https://github.com/balena-io/etcher/issues/414).
As you can see even in that issue though, the scope of Etcher right now is rather the one of flashing without modifying the source image. This though might be something we want to explore once we start fiddling with balenaOS images configuration so we might take it from there

thundron, 2019-01-16:
> a "generic" way to convert a "raw .img" into a "bootable .iso" - AFAIK such a thing doesn't exist as it would need lots of image-specific knowledge about what exactly is in the image, what extra files need to be added to make it bootable as an ISO (e.g. https://www.syslinux.org/wiki/index.php?title=ISOLINUX ), etc. etc.
This, which is the reason that "for now" we're not considering it.
It might be a nice addition in the future, who knows

pcgeek86, 2019-01-16:
That makes sense, thanks for the additional context.

My imagination was that you could mount a raw .img using a loopback interface (`mount -o loop`) and then run `mkisofs` to generate a bootable `.iso` file. I see what you mean about there not being a generic method of taking a disk image, with an unknown partition scheme, and merging it into an ISO. That definitely justifies putting this over into the balenaOS repository, such that a bootable ISO could be purpose-built from scratch instead.

On a side note, there's a [closed-source Windows tool](http://www.imgtoiso.com) that claims to convert .img to .iso files, but it's unclear exactly how this works. I'm not inclined to test it without sandboxing it or running it on a disposable OS.

https://unix.stackexchange.com/questions/316401/how-to-mount-a-disk-image-from-the-command-line
https://linux.die.net/man/8/mkisofs

lurch, 2019-01-16:
> That makes sense, thanks for the additional context.

No problem.

> there's a closed-source Windows tool that claims to convert .img to .iso files, but it's unclear exactly how this works.

That page says "The IMG is an disk image file format, it be used by some virtual drive software" so it probably just assumes that the IMG contains a single partition (probably formatted to NTFS), which it then mounts and copies all the files into an ISO9660 filesystem. Much like your combination of `mount -o loop` and `mkisofs` :slightly_smiling_face: 

In case you've not realised, ISO files aren't "automatically bootable", they need special boot code on them; just the same way as a USB stick with a bunch of regular data files on it isn't "bootable". It's just that in this day & age most of the ISOs you download are setup to be bootable (e.g. as OS installers or LiveLinux distros).

thundron, 2019-01-16:
Closing for now, will reopen/create a new and more verbose issue if needed

closed 2019-01-16 by thundron