balena-io/etcher #2605: Etcher has been separate partition on USB drive without make it bootable

KittipatGH, 2019-01-16:
- **Etcher version: 1.4.9** 
- **Operating system and architecture: Windows 10  x64** 
- **Image flashed: Bootable Chrome OS** 
- **Do you see any meaningful error information in the DevTools? : No, I didn't see anything about error I see the result after making bootable drive with Etcher**

<u>Normally after I used other tools to make bootable drive, after finished I'll see the nice results that's it containing with selected image file, but Etcher is not, don't just separate partition on my drive but not containing anything that's I needed, Please look this picture in Removable drive with 7.21 GB can be use, I won't use this until It have been fixed with other bugs as others created issues.</u>
[image]

lurch, 2019-01-16:
That picture looks totally normal - Etcher is just a "raw" flasher tool (until #413 gets implemented), and so the sizes and layouts of the partitions you get after flashing an image to your drive are determined entirely by whatever is contained in the image.

thundron, 2019-01-16:
Please refer to [this](https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#why-is-my-drive-not-bootable) for bootable drives. Etcher doesn't support that unless the image itself is specifically made to do so. Closing

closed 2019-01-16 by thundron