balena-io/etcher #2612: Failure adding etcher repository

mfbalin, 2019-01-25:
- **Operating system and architecture:** Ubuntu 18.04 

Failure when trying to add etcher repository to apt.

Command: sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 379CE192D401AB61
Output: gpg: keyserver receive failed: Server indicated a failure

lurch, 2019-01-25:
Maybe a temporary glitch? Working for me.

mfbalin, 2019-01-25:
Is there any other way of signing the repository?

thundron, 2019-01-31:
@mfbalin You can try using other mirrors if `keyserver.ubuntu.com` doesn't work for you, here's a list with [some of them](https://www.rossde.com/PGP/pgp_keyserv.html#pubserv)

camilojm27, 2019-02-03:
Do not work  for me neither i am using deepin (debian based)
[image]

thundron, 2019-05-06:
@camilojm27 Did you try the other keyservers listed at https://www.rossde.com/PGP/pgp_keyserv.html#pubserv by chance?

brianleePTL, 2019-05-30:
I'm still getting this error. I have tried the other keyservers and still get the same thing. Does anyone have a verified keyserver that they know works?

tilsammans, 2019-10-17:
This is a DNS issue. To fix it, make sure you are using a DNS resolver that doesn't filter.

e.g. in `/etc/resolv.conf` have one of these:
```
nameserver 8.8.8.8 # Google public DNS
nameserver 1.1.1.1 # or Cloudflare
```

thundron, 2019-10-21:
Thanks @tilsammans , closing this one as it's (unfortunately) unrelated to Etcher itself

closed 2019-10-21 by thundron