balena-io/etcher #2614: Unexpected token x in JSON at position 5100

epsilonrt, 2019-01-28:
- Etcher Gui 1.4.9 
- Ubuntu Xenial amd64

[image]
This window always appears at startup.

thundron, 2019-01-29:
@epsilonrt Could you please paste here the output of `lsblk --all --json --paths --output-all`?

epsilonrt, 2019-01-29:
Hi,
`
lsblk --all --json --paths --output-all
{
   "blockdevices": [
      {"name": "/dev/sda", "kname": "/dev/sda", "maj:min": "8:0", "fstype": null, "mountpoint": null, "label": null, "uuid": null, "parttype": null, "partlabel": null, "partuuid": null, "partflags": null, "ra": "128", "ro": "0", "rm": "0", "hotplug": "0", "model": "ST500LM000-1EJ16", "serial": "W76570S8", "size": "465,8G", "state": "running", "owner": "root", "group": "disk", "mode": "brw-rw----", "alignment": "0", "min-io": "4096", "opt-io": "0", "phy-sec": "4096", "log-sec": "512", "rota": "1", "sched": "deadline", "rq-size": "128", "type": "disk", "disc-aln": "0", "disc-gran": "0B", "disc-max": "0B", "disc-zero": "0", "wsame": "0B", "wwn": "0x5000c5008a6ce52c", "rand": "1", "pkname": null, "hctl": "0:0:0:0", "tran": "sata", "subsystems": "block:scsi:pci", "rev": "SM36", "vendor": "ATA     ",
         "children": [
            {"name": "/dev/sda1", "kname": "/dev/sda1", "maj:min": "8:1", "fstype": "ntfs", "mountpoint": null, "label": "Win7x64", "uuid": "D0A43186A4316FD8", "parttype": "0x7", "partlabel": null, "partuuid": "849eafe5-01", "partflags": "0x80", "ra": "128", "ro": "0", "rm": "0", "hotplug": "0", "model": null, "serial": null, "size": "138G", "state": null, "owner": "root", "group": "disk", "mode": "brw-rw----", "alignment": "0", "min-io": "4096", "opt-io": "0", "phy-sec": "4096", "log-sec": "512", "rota": "1", "sched": "deadline", "rq-size": "128", "type": "part", "disc-aln": "0", "disc-gran": "0B", "disc-max": "0B", "disc-zero": "0", "wsame": "0B", "wwn": "0x5000c5008a6ce52c", "rand": "1", "pkname": "/dev/sda", "hctl": null, "tran": null, "subsystems": "block:scsi:pci", "rev": null, "vendor": null},
            {"name": "/dev/sda3", "kname": "/dev/sda3", "maj:min": "8:3", "fstype": null, "mountpoint": null, "label": null, "uuid": null, "parttype": "0x5", "partlabel": null, "partuuid": "849eafe5-03", "partflags": null, "ra": "128", "ro": "0", "rm": "0", "hotplug": "0", "model": null, "serial": null, "size": "1K", "state": null, "owner": "root", "group": "disk", "mode": "brw-rw----", "alignment": "1024", "min-io": "4096", "opt-io": "0", "phy-sec": "4096", "log-sec": "512", "rota": "1", "sched": "deadline", "rq-size": "128", "type": "part", "disc-aln": "0", "disc-gran": "0B", "disc-max": "0B", "disc-zero": "0", "wsame": "0B", "wwn": "0x5000c5008a6ce52c", "rand": "1", "pkname": "/dev/sda", "hctl": null, "tran": null, "subsystems": "block:scsi:pci", "rev": null, "vendor": null},
            {"name": "/dev/sda4", "kname": "/dev/sda4", "maj:min": "8:4", "fstype": "ext4", "mountpoint": "/", "label": null, "uuid": "60682565-e840-4cce-a7ec-e16c4e0dfdc0", "parttype": "0x83", "partlabel": null, "partuuid": "849eafe5-04", "partflags": null, "ra": "128", "ro": "0", "rm": "0", "hotplug": "0", "model": null, "serial": null, "size": "100,5G", "state": null, "owner": "root", "group": "disk", "mode": "brw-rw----", "alignment": "0", "min-io": "4096", "opt-io": "0", "phy-sec": "4096", "log-sec": "512", "rota": "1", "sched": "deadline", "rq-size": "128", "type": "part", "disc-aln": "0", "disc-gran": "0B", "disc-max": "0B", "disc-zero": "0", "wsame": "0B", "wwn": "0x5000c5008a6ce52c", "rand": "1", "pkname": "/dev/sda", "hctl": null, "tran": null, "subsystems": "block:scsi:pci", "rev": null, "vendor": null},
            {"name": "/dev/sda5", "kname": "/dev/sda5", "maj:min": "8:5", "fstype": "ext4", "mountpoint": null, "label": null, "uuid": "17910e87-41b9-4e09-9489-d49a3eb64038", "parttype": "0x83", "partlabel": null, "partuuid": "849eafe5-05", "partflags": null, "ra": "128", "ro": "0", "rm": "0", "hotplug": "0", "model": null, "serial": null, "size": "47,7G", "state": null, "owner": "root", "group": "disk", "mode": "brw-rw----", "alignment": "0", "min-io": "4096", "opt-io": "0", "phy-sec": "4096", "log-sec": "512", "rota": "1", "sched": "deadline", "rq-size": "128", "type": "part", "disc-aln": "0", "disc-gran": "0B", "disc-max": "0B", "disc-zero": "0", "wsame": "0B", "wwn": "0x5000c5008a6ce52c", "rand": "1", "pkname": "/dev/sda", "hctl": null, "tran": null, "subsystems": "block:scsi:pci", "rev": null, "vendor": null}
         ]
      },
      {"name": "/dev/sdb", "kname": "/dev/sdb", "maj:min": "8:16", "fstype": null, "mountpoint": null, "label": null, "uuid": null, "parttype": null, "partlabel": null, "partuuid": null, "partflags": null, "ra": "128", "ro": "0", "rm": "0", "hotplug": "0", "model": "ST2000LM003 HN-M", "serial": "S377J9AG900162", "size": "1,8T", "state": "running", "owner": "root", "group": "disk", "mode": "brw-rw----", "alignment": "0", "min-io": "4096", "opt-io": "0", "phy-sec": "4096", "log-sec": "512", "rota": "1", "sched": "deadline", "rq-size": "128", "type": "disk", "disc-aln": "0", "disc-gran": "0B", "disc-max": "0B", "disc-zero": "0", "wsame": "0B", "wwn": "0x50004cf210bb739d", "rand": "1", "pkname": null, "hctl": "1:0:0:0", "tran": "sata", "subsystems": "block:scsi:pci", "rev": "0001", "vendor": "ATA     ",
         "children": [
            {"name": "/dev/sdb1", "kname": "/dev/sdb1", "maj:min": "8:17", "fstype": "ntfs", "mountpoint": "/media/Donn\xc3\xa9es", "label": "Donn\xc3\xa9es", "uuid": "ECFA6CAEBEC71B44", "parttype": "0x7", "partlabel": null, "partuuid": "c3a816a5-01", "partflags": null, "ra": "128", "ro": "0", "rm": "0", "hotplug": "0", "model": null, "serial": null, "size": "1000G", "state": null, "owner": "root", "group": "disk", "mode": "brw-rw----", "alignment": "0", "min-io": "4096", "opt-io": "0", " [... cut ...] , "hotplug": "0", "model": null, "serial": null, "size": null, "state": null, "owner": "root", "group": "disk", "mode": "brw-rw----", "alignment": "0", "min-io": "512", "opt-io": "0", "phy-sec": "512", "log-sec": "512", "rota": "1", "sched": null, "rq-size": "128", "type": "loop", "disc-aln": "0", "disc-gran": "0B", "disc-max": "0B", "disc-zero": "0", "wsame": "0B", "wwn": null, "rand": "0", "pkname": null, "hctl": null, "tran": null, "subsystems": "block", "rev": null, "vendor": null}
   ]
}

`

lurch, 2019-01-29:
FWIW the text above gets parsed fine by Python's JSON pasrser, but https://jsonlint.com/ complains about `"mountpoint": "/media/Donn\xc3\xa9es"` - looks like a classic Unicode escaping / unescaping issue.

epsilonrt, 2019-01-29:
Hi,
Bingo, thanks a lot !
I changed the mount point name and partition label to remove the UTF-8 character and it works.

closed 2019-01-29 by epsilonrt

lurch, 2019-01-30:
...although this is probably still a bug (in `drivelist`) that needs fixing, in case anyone else runs into it in future?

zvin, 2019-01-31:
Blame it on lsblk https://github.com/karelzak/util-linux/issues/330