balena-io/etcher #2617: Windows 7 64 bits : Permission error

Nickduino, 2019-01-31:
- **Etcher version:** 1.4.9
- **Operating system and architecture:** Windows 7 64 bits
- **Image flashed:** 2018-11-13-raspbian-stretch-lite.zip
- **Do you see any meaningful error information in the DevTools?** Yes (attached below)

The error message (I tried launching from a command line in case it displayed a feedback):

[image]

The devtool info:
[image]

5 minutes ago, I was using this very same SD in my Raspberry Pi. I'm not saying it's not broken...but it seemed to be working then.

thundron, 2019-01-31:
@Nickduino Did you try running it as administrator too?

Nickduino, 2019-01-31:
I should have mentionned but yes I did

Nickduino, 2019-01-31:
I'm using the SD card that came with my Raspberry Pi. Is it possible it is somewhat write protected?

thundron, 2019-01-31:
It sounds strange, but it looks like [it happened already before](https://github.com/balena-io/etcher/issues/1426#issuecomment-386395975).
You might want to look into other tools to see if this happens there too, according to [this issue](https://github.com/pbatard/rufus/issues/313#issuecomment-40554298) as well

snow8gaia, 2019-01-31:
Same issue.

- **Etcher version:** balenaEtcher-Portable-1.4.9-x64.exe
- **Operating system and architecture:** Windows 10 1809 64-bit 
- **Image flashed:** FWUL_v3.1_x86_64_forgetful.iso ,archlinux-2019.01.01-x86_64.iso, kubuntu-18.10-desktop-amd64.iso
**Reproducible:** 5 out of 5.
- **Do you see any meaningful error information in the DevTools?:**   (kubuntu iso)

# Yes
<details>
<summary>
copy pasted logs, Click to expand
</summary>

```javascript
Umount on success: true

Validate on success: true

Fri Feb 01 2019 01:01:44 GMT+0530 (India Standard Time) Flash error ({"error":{"name":"Error","message":"EPERM: operation not permitted, write","stack":"Error: EPERM: 
operation not permitted, write","code":"EPERM","syscall":"write","errno":-4048},"applicationSessionUuid":"a231da97-d139-4907-
9027-4fa24541105f","flashingWorkflowUuid":"2b07360d-f80a-44e1-9cce-1bc2283674dd","flashInstanceUuid":"6fa420ea-fae1-4064-8e85-0703b50d3e97","image":"F:\\Backup programs\\Linux OS for VM\\ISO files\\ubuntu
 family\\kubuntu\\18.10\\kubuntu-18.10-desktop-amd64.iso","drives":["\\\\.\\PhysicalDrive2"],"driveCount":1,"uuid":"6fa420ea-fae1-4064-8e85-0703b50d3e97","unmountOnSuccess":true,"validateWriteOnSuccess":true})

Error: EPERM: operation not permitted, write

Fri Feb 01 2019 01:01:44 GMT+0530 (India Standard Time) Open modal ({"name":"warning","applicationSessionUuid":"a231da97-d139-4907-9027-4fa24541105f","flashingWorkflowUuid":"2b07360d-f80a-44e1-9cce-1bc2283674dd"})

{"stack":"Error: EPERM: operation not permitted, write","message":"EPERM: operation not permitted, write","name":"Error","code":"EPERM","syscall":"write","errno":-4048}
console.(anonymous function) @ C:\Users\trtsqwe\AppData\Local\Temp\nsz8224.tmp\app\resources\app.asar\node_modules\raven-js\dist\rave…:58

etcher:sdk:scanner start +10s true

Warning: Failed prop type: Invalid prop `disabled` of type `string` supplied to `ProgressButton`, expected `boolean`.
    in ProgressButton
console.(anonymous function) @ C:\Users\trtsqwe\AppData\Local\Temp\nsz8224.tmp\app\resources\app.asar\node_modules\raven-js\dist\rave…:58

Flash results Object                             
                              cancelled: false
                              __proto__: Object                            

Terminating IPC server
```
</details>

 I used USB 3.0 port. Ran as admin.

**How I handled it:**

1. Press Win key.
2. Type:- disk management
3. Click on Create and format Hard disk partitions.
4. Disk Management opens.
5. There is no partition on flash drive. The previous partition got removed. 
6. Right click on disk and select 'New Simple Volume'. Format it as per requirements, eg: FAT32, quick format.
7. Click Next> Finish. Wait for a while.
For more methods:-  [https://fossbytes.com](https://fossbytes.com/how-to-repair-corrupted-pen-drive-or-sd-card/)

I think Etcher got interrupted by Windows Security's  'Controlled folder access' feature.
1. Press Win key.
2. Type:- Controlled folder access
3. Click on Controlled folder access and configure the settings. Apps can also be allowed access to folders.
Note:- I guess by default folders in C:\Users\username , C:\Users\Public are protected. So ,Documents, Pictures, Favorites, Videos, etc.

# the images:

<details>
<summary>
Click to expand
</summary><p>

etcher error message

[image]

pen drive size after etcher

[image]

windows security CFA notifications

[image]

</p>
</details>

I hope it helps.

snow8gaia, 2019-01-31:
could be similar https://github.com/balena-io/etcher/issues/2547

I'll be unavailable for quick responses for a while.

Nickduino, 2019-02-02:
It did work flawlessly with a brand new Kingston.

I'm convinced the other SD had some kind of write protection (not allowing to change the recovery partition)

closed 2019-02-02 by Nickduino

snow8gaia, 2019-02-03:
Taking https://github.com/balena-io/etcher/issues/2617#issuecomment-459961487  into account, this becomes a task for improving experience on Windows 10. If I need to file a separate issue, please let me know.

If yes:
- I think the title could be 'Windows 10 1809 64 bits : Permission error( possibly due to security settings )'. Improvements welcome. 
-   Should I copy the same info over to the new issue or just a link will do? 

I will try to see if I can find any more security settings/policies that stump Etcher. And try alternatives like Rufus.

snow8gaia, 2019-02-03:
Rufus is out.

I have read https://github.com/pbatard/rufus/wiki/FAQ#error-0x00000005-access-is-denied , but is it true for Etcher?

(this should be easy to find) https://github.com/pbatard/rufus/wiki/FAQ#incompatible-software, is there a similar section for Etcher?

snow8gaia, 2019-02-03:
My bad, I tried Rufus( portable version ) without changing my Anti-Ransomware( CFA ) settings, and it succeeded where Etcher( portable version ) failed. 

So, if Etcher fails and CFA is needed, try Rufus. It may work.

Nickduino, 2019-02-03:
I have to admit I don't follow you :-D

What are you saying? That Rufus might be able to repartition par (it seems) partition-locked SD card?

snow8gaia, 2019-02-04:
No, I used a flash drive.

snow8gaia, 2019-02-04:
About filing a new issue:

>-   Should I copy the same info over to the new issue or just a link will do?