balena-io/etcher #2618: Error in dialog box on opening Mac app

jywarren, 2019-01-31:
- **Etcher version:** Version 1.4.4 (1.4.4)
- **Operating system and architecture:** Mac OS 10.13.6 (17G4015)
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?** 

Just got a popup with an error as follows on opening the app:

[image]

```
TypeError: Cannot read property 'UPDATE_NOTIFIER_SLEEP_DAYS' of undefined
    at Object.exports.notify (file:///Applications/Etcher.app/Contents/Resources/app.asar/generated/gui.js:118076:55)
    at file:///Applications/Etcher.app/Contents/Resources/app.asar/generated/gui.js:65181:27
    at tryCatcher (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/util.js:16:23)
    at Promise._settlePromiseFromHandler (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:504:31)
    at Promise._settlePromise (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:561:18)
    at Promise._settlePromise0 (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:606:10)
    at Promise._settlePromises (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:685:18)
    at Promise._fulfill (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:630:18)
    at Promise._resolveCallback (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:425:57)
    at Promise._settlePromiseFromHandler (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:516:17)
    at Promise._settlePromise (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:561:18)
    at Promise._settlePromise0 (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:606:10)
    at Promise._settlePromises (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:685:18)
    at Promise._fulfill (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:630:18)
    at MappingPromiseArray.PromiseArray._resolve (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise_array.js:125:19)
    at MappingPromiseArray._filter (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/map.js:123:10)
    at MappingPromiseArray._promiseFulfilled (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/map.js:95:18)
    at MappingPromiseArray.PromiseArray._iterate (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise_array.js:113:31)
    at MappingPromiseArray.init (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise_array.js:77:10)
    at Promise._settlePromise (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:558:21)
    at Promise._settlePromise0 (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:606:10)
    at Promise._settlePromises (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:685:18)
    at Promise._fulfill (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:630:18)
    at Promise._settlePromise (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:574:21)
    at Promise._settlePromise0 (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:606:10)
    at Promise._settlePromises (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:685:18)
    at Promise._fulfill (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:630:18)
    at Promise._resolveCallback (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:425:57)
    at Promise._settlePromiseFromHandler (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:516:17)
    at Promise._settlePromise (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:561:18)
    at Promise._settlePromise0 (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:606:10)
    at Promise._settlePromises (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:685:18)
    at Promise._fulfill (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:630:18)
    at Promise._resolveCallback (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:425:57)
    at ReductionPromiseArray._resolve (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/reduce.js:51:19)
    at Promise.completed [as _fulfillmentHandler0] (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/reduce.js:112:15)
    at Promise._settlePromise (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:558:21)
    at Promise._settlePromise0 (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:606:10)
    at Promise._settlePromises (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:685:18)
    at Async._drainQueue (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/async.js:138:16)
    at Async._drainQueues (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/async.js:148:10)
    at Async.drainQueues (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/async.js:17:14)
    at <anonymous>
    at process._tickCallback (internal/process/next_tick.js:109:7)
```

lurch, 2019-01-31:
I believe this error has already been fixed in newer versions of Etcher.

thundron, 2019-01-31:
It has, it was the update notification, fixed in 1.4.9. Closing