balena-io/etcher #2619: Etcher 1.4.9 (CLI) can't find USB drive

GGShinobi, 2019-02-03:
- **Etcher version:** 1.4.9 CLI
- **Operating system and architecture:**
```
sh-3.2# system_profiler SPSoftwareDataType
Software:

    System Software Overview:

      System Version: Mac OS X 10.7.5 (11G63)
      Kernel Version: Darwin 11.4.2
      Boot Volume: Mac HD
      Boot Mode: Normal
      Computer Name: Selinas MacBook
      User Name: System Administrator (root)
      Secure Virtual Memory: Enabled
      64-bit Kernel and Extensions: No
      Time since boot: 55 minutes

sh-3.2# system_profiler SPHardwareDataType
Hardware:

    Hardware Overview:

      Model Name: MacBook
      Model Identifier: MacBook3,1
      Processor Name: Intel Core 2 Duo
      Processor Speed: 2,2 GHz
      Number of Processors: 1
      Total Number of Cores: 2
      L2 Cache: 4 MB
      Memory: 3 GB
      Bus Speed: 800 MHz
      Boot ROM Version: MB31.008E.B02
      SMC Version (system): 1.24f3
      Sudden Motion Sensor:
          State: Enabled

```
`
- **Image flashed:** N/A (tried: lubuntu-18.04.1-desktop-amd64.iso)

etcher can't find the drive, even though it's definitely mounted:

```
sh-3.2# mount
/dev/disk0s2 on / (hfs, local, journaled)
devfs on /dev (devfs, local, nobrowse)
map -hosts on /net (autofs, nosuid, automounted, nobrowse)
map auto_home on /home (autofs, automounted, nobrowse)
/dev/disk1s2 on /Volumes/MACBOOT (msdos, local, nodev, nosuid, noowners)
sh-3.2# etcher -v
1.4.9
sh-3.2# etcher -d /dev/disk1s2 lubuntu-18.04.1-desktop-amd64.iso 
? This will erase the selected drive. Are you sure? Yes
The selected drive "/dev/disk1s2" was not found

We can't find "/dev/disk1s2" in your system. Did you unplug the drive?
sh-3.2# 

```

I also tried:

```
sh-3.2# etcher -d MACBOOT lubuntu-18.04.1-desktop-amd64.iso
? This will erase the selected drive. Are you sure? Yes
The selected drive "MACBOOT" was not found

We can't find "MACBOOT" in your system. Did you unplug the drive?
sh-3.2# etcher -d /Volumes/MACBOOT lubuntu-18.04.1-desktop-amd64.iso
? This will erase the selected drive. Are you sure? Yes
The selected drive "/Volumes/MACBOOT" was not found

We can't find "/Volumes/MACBOOT" in your system. Did you unplug the drive?

```

lurch, 2019-02-04:
I believe you need to give it the disk name (i.e. `/dev/disk1`) rather than the partition name (`/dev/disk1s2`) ?

@thundron ISTR somebody else stumbling across this before - maybe it's worth adding a specific check and warning message for? (i.e. user specifying a partition instead of a disk)

GGShinobi, 2019-02-04:
> I believe you need to give it the disk name (i.e. `/dev/disk1`) rather than the partition name (`/dev/disk1s2`) ?
> 
> @thundron ISTR somebody else stumbling across this before - maybe it's worth adding a specific check and warning message for? (i.e. user specifying a partition instead of a disk)

Thank you very much!!!! I tried that and it worked!!! :+1:  It's burning the image right now!!! :)

thundron, 2019-02-04:
@lurch Yes it's probably worth adding a note there once we ship Electron 3 + NPM (soon™)

thundron, 2019-04-08:
The CLI has been discontinued, we now support the desktop application only

closed 2019-04-08 by thundron