balena-io/etcher #2623: etcher to debian main repo

sumanrajan, 2019-02-07:
First of all i want to thank the devs for such a cool software 
Secondly i want to see etcher in debian main repos 
If i could get some help i would like to volunter to it as well 
Hope the dev team will look into it 
Thankyou

lurch, 2019-02-08:
The Etcher team already provide a `.deb` file ( https://github.com/balena-io/etcher#debian-and-ubuntu-based-package-repository-gnulinux-x86x64 ) - I think to get it in the "debian main repos" is an issue for Debian, rather than an issue for Balena? :man_shrugging:

camilojm27, 2019-02-10:
> The Etcher team already provide a `.deb` file ( https://github.com/balena-io/etcher#debian-and-ubuntu-based-package-repository-gnulinux-x86x64 ) - I think to get it in the "debian main repos" is an issue for Debian, rather than an issue for Balena? man_shrugging
> 
> EDIT: To clarify, the opinions are my own and I'm not involved in any way with Balena! (or Debian)

The problem is the gpt signature, the server is down, i have to use an appImagen

lurch, 2019-02-10:
> The problem is the gpt signature, the server is down

@thundron provided a (link to a) list of keyserver mirrors in #2612

thundron, 2019-02-16:
Going to integrate that list in the readme soon™

closed 2019-02-16 by thundron