balena-io/etcher #2625: Etcher live USB is freezing laptop restart

DevAspirerKlaus, 2019-02-11:
- **Etcher version: 1.4.9** 
- **Operating system and architecture: Windows 10 Single Language** 
- **Image flashed: Pop! OS (but I tested with other images - same thing)** 
- **Do you see any meaningful error information in the DevTools?: No error at all** 
[image]

Just after flashed the ISO, the USB stick acts normally. Ejecting and putting it again does nothing unusual with the OS running. But as soon as I restart and the screen show the brand screen, when I press F2 or F12, the system freezes. With F2, if I remove the USB stick, it lets the laptop enter the BIOS. Inside it, if I plug it again, it freezes. Same when you press F12. It only passes to the boot menu when I plug it off. If I try to plug it on, it freezes again.

I'm having some issues with other flashing software, so my last hope is Etcher working good.
,

lurch, 2019-02-12:
Sounds like a problem (bug?) with your computer's BIOS? Which Etcher obviously can't do much about...

Might be worth checking if the manufacturer of your laptop has a BIOS update available?

DevAspirerKlaus, 2019-02-12:
Thanks for answering, @lurch. I'm trying to flash with a previous version of Etcher. I remember that a few months ago I didn't have this kind of problem. Let's see what will happen...

DevAspirerKlaus, 2019-02-16:
Well, I will close this issue as it seems the way Etcher flashes data freezes my specific laptop (truly other flashing softwares, like UNetbootin, are doing good) does not trigger this issue on anybody else. If someone willing to have more information about it (if this is happening with them), they can comment here and I'll help :)

closed 2019-02-16 by DevAspirerKlaus

lurch, 2019-02-16:
It's not an Etcher-specific problem, if you used `dd` or `Win32DiskImager` to write your images you'd have exactly the same problem. Unetbootin actually does something entirely different - see #413 for more info.
Perhaps something like https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos is what's confusing your laptop's BIOS? :man_shrugging:

DevAspirerKlaus, 2019-02-17:
@lurch thanks. I will ready those articles and see if a light bulb pops in my head.