balena-io/etcher #2627: no module found "../generated/etcher"

an0nym0u5-404, 2019-02-12:
- **Etcher version:** 
- **Operating system and architecture:** ubuntu 16.04
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?** 

 Hi I am new to electron and etcher, and I have tried to build the tool from source by "npm install" for installing dependencies. However, when I run "npm start" which basically calls "electron lib/start.js", it tells me missing module "../generated/etcher" which is caused by the require(...) statement in start.js. Therefore, I am just wondering what is the correct way of building the tool from source if I have done anything wrong? Please let me know and thank you!

thundron, 2019-02-12:
@an0nym0u5-404 Before that, you need to run `make webpack` or if you are trying to develop and want to continuously build the UI, `npm run watch` watches changes to javascript files and updates the generated bundle accordingly

an0nym0u5-404, 2019-02-12:
@thundron thank you so much!

lurch, 2019-02-13:
You can't use "npm install" with Etcher, you need to read through https://github.com/balena-io/etcher/blob/master/docs/CONTRIBUTING.md#developing :wink:  (although I've no idea if those instructions are still up to date)

closed 2019-02-16 by thundron