balena-io/etcher #2633: Getting Plain Screen On Ubuntu 18.04 after following the steps given in guidelines..

shivarajnaidu, 2019-02-15:
- **Etcher version:**  : Cloned From Latest Master
- **Operating system and architecture:**  
Distributor ID: Ubuntu
Description:    Ubuntu 18.04.2 LTS
Release:        18.04
Codename:       bionic

- **Do you see any meaningful error information in the DevTools?** 

[image]

[image]

thundron, 2019-02-15:
@shivarajnaidu Did you run `make electron-develop` too?

shivarajnaidu, 2019-02-15:
Yes @thundron .. got this error
[image]

thundron, 2019-02-15:
@shivarajnaidu Run `git submodule init && git submodule update`, then retry `make electron-develop`

shivarajnaidu, 2019-02-15:
Getting error.. for this too.. 
[image]

shivarajnaidu, 2019-02-15:
Got this error.. 
[image]

shivarajnaidu, 2019-02-15:
After installing jq .. it's compiling properly

thundron, 2019-02-15:
@shivarajnaidu Follow the guide [here](https://github.com/balena-io/etcher/blob/master/docs/CONTRIBUTING.md) to check that you have everything

shivarajnaidu, 2019-02-15:
Sure.. I followed the same @thundron  It's working now
Thank You very much.. 

finally it gets compiled successfully ..
 I think we can add 
Run `git submodule init && git submodule update`, then retry `make electron-develop` these steps and `sudo apt install jq` command in linux section.. 
I think it will be helpful for new contributors. Please consider adding these in contribution guide. Thank You Once Again.

thundron, 2019-02-15:
@shivarajnaidu Definitely, and remember you can always issue a PR for that as well! They are always very welcome and encouraged

closed 2019-02-15 by thundron

shivarajnaidu, 2019-02-15:
Sure.. I'll try to rise an PR as soon as possible.. Thank You for your kind and timely guidance @thundron  :relaxed: