balena-io/etcher #2636: Libfuse error on Kubuntu 18.10

vincegre, 2019-02-17:
- Etcher version: 1.4.9
- Operating system and architecture: Kubuntu 18.10 64 bits
- Do you see any meaningful error information in the DevTools?Nope as unable to start it but here is what I get in console if starting it from command line:

```
vincen@fixe:~$ chmod a+x balena-etcher-electron-1.4.9-i386.AppImage 
vincen@fixe:~$ ./balena-etcher-electron-1.4.9-i386.AppImage 
./balena-etcher-electron-1.4.9-i386.AppImage: error while loading shared libraries: libfuse.so.2: cannot open shared object file: No such file or directory
vincen@fixe:~$
```

Side note: why site offers me only 32 bit download for Linux 64 bits I'm running on ??

lurch, 2019-02-17:
Sounds like a duplicate of #1490 ?

vincegre, 2019-02-17:
Ah yep sorry for noise !

closed 2019-02-17 by vincegre