balena-io/etcher #2638: Request to change "unusually large for an SD card or USB stick" threshold

ras07, 2019-02-21:
Ver. 1.4.9 warns that the device is "unusually large for an SD card or USB stick" if the device is 64 GB or larger. However 64 GB cards are increasingly common, and can currently be had on Amazon (from reliable vendors) for under $10. (It hardly even makes sense to buy a smaller card anymore, since a 32 GB card is only about $2 less.) So the "false positive" rate for this warning (when it warns you even though you have selected the intended device) is getting high.

As 64GB+ cards are rapidly becoming the norm, people will get used to ignoring this warning. This will have the inevitable result that some users will accidentally flash unintended devices, because they are used to clicking Continue when they see this warning. (I, for one, can totally see myself doing this.)

125GB cards aren't far behind at under $25 and dropping, so I would therefore propose increasing the "unusually large" threshold to 256 GB or perhaps even higher.

lurch, 2019-02-21:
IIRC this warning was added because people were accidentally using Etcher to overwrite all the files they had on a USB HDD (and consequently getting very upset). So it's probably always going to be a bit of a balancing act to decide where this limit should be set :confused: 

@thundron Maybe it's worth adding an environment-variable that people can set if they want to increase the size of the disk that Etcher will consider "unusually large"?

lightmaster, 2019-11-26:
Would it be possible to set it to warn when an SD card is 256GB or larger, or if a USB flash drive is 64GB or larger?

People most commonly use MicroSD cards for their RPis and similar devices, but a detected USB drive is much less likely to be the intended target.

lurch, 2019-12-02:
Unfortunately there's no reliable way to differentiate between "an SD card" and a "USB flash drive" (especially as there's such a huge variety of SD card readers) - in many cases they both just appear as a "USB Mass Storage Device" to the Operating System.

grerrg, 2019-12-28:
We are at the point were 200GB microSD cards are almost the cheapest GB/per currency.

[image]

The prices are halving each year for the same capacity. 400GB will be common in one year and 800 the following year. A threshold at one TB seams fine for the future.

Even today it doesn't make sense buying a 1TB HDD, since 2TB are cheaper.

lurch, 2021-04-08:
Looks like this threshold is now 128GB ?
https://github.com/balena-io/etcher/blob/master/lib/shared/drive-constraints.ts#L145

Jubei-VA, 2024-06-04:
Would be nice if it was set slightly higher now.