balena-io/etcher #2639: Inaccurate info about macOS requirements (docs/CONTRIBUTING.md)

zrzka, 2019-02-21:
The documentation says ...

> [XCode](https://developer.apple.com/xcode/) or [XCode Command Line Tools](https://developer.apple.com/library/content/technotes/tn2339/_index.html), which can be installed by running `xcode-select --install`.

... but when I try to run `make electron-develop`, it fails to build `drivelist` with the following error ...

```
> drivelist@7.0.3 install /Users/robertvojta/Work/balena/etcher/node_modules/drivelist
> prebuild-install || node-gyp rebuild

xcode-select: error: tool 'xcodebuild' requires Xcode, but active developer directory '/Library/Developer/CommandLineTools' is a command line tools instance
```

... which means that Xcode must be installed, not just Command Line Tools.

zrzka, 2019-02-21:
Just installed Xcode and can confirm that it works as expected now.

lurch, 2019-02-21:
Perhaps that's related to https://github.com/balena-io-modules/drivelist/pull/324 ? :man_shrugging:

zrzka, 2019-02-22:
@lurch why do you think it's related? I'm not JS expert, just tried to build it on macOS and it failed. I  opened this PR, because I think that it's safer to say that the Xcode is required (will install Xcode command line tools when you run it) than just Xcode command line tools. In case prebuild drivelist won't be available, you'll probably need Xcode anyway. But as I said, I may be wrong and you meant something else I didn't understand.

lurch, 2019-02-22:
Sorry if I wasn't clear - I meant that maybe that change to drivelist means that "Xcode command line tools" is no longer sufficient (when it used to be when the instructions here were first written), and now "full Xcode" is required instead.

Apologies for any confusion caused :slightly_frowning_face:

zrzka, 2019-02-22:
Ah, I've got it now, thanks for the clarification 👍

commit 63c8cd21 (2019-02-21): Update macOS contributing info

Fixes #2639

* Fixed typos (OS X -> macOS, XCode -> Xcode)
* Xcode is required, not just Xcode Command Line Tools

Change-type: patch
closed 2023-07-12 by dfunckt