balena-io/etcher #2641: Missing Dependency in AppImage: libgconf2-4

matrixes, 2019-02-22:
- **Etcher version:** 
1.4.9

- **Operating system and architecture:** 
Ubuntu 18.04.2 - Desktop edition - x64  - FAIRLY FRESH install

- **Image flashed:** 
None

- **Do you see any meaningful error information in the DevTools?** 
No

- **What is the problem you're experiencing? If possible, provide a list of steps you performed**

1. Downloaded v1.4.9 from here https://www.balena.io/etcher/
2. Extracted the AppImage after download
3. Double clicked the AppImage and got a question to install a shortcut (or something similar to that), I answered yes
4. Opened Activities, started typing etcher and selected the balenaEtcher item in the results by pressing enter
5. Nothing
6. Absolutely nothing
7. Not even an error message
8. Searched the internet and found a hint on installing libgconf2-4 in this issue https://github.com/balena-io/etcher/issues/1305

Do AppImages prevent the developers making them, to provide error messages or any useful information at all as to why an application won't launch?

(I couldn't find anything but previously closed issues on this, so I guess I'm supposed to create new issues?)

lurch, 2019-02-22:
It's also worth checking **which** AppImage you've downloaded - I'm not sure why but it seems quite a few people have been attempting to run the 32-bit (i386) AppImage on 64-bit (x86_64) Linux distros, which won't work.

matrixes, 2019-02-25:
There you go, updated the issue description. As I'm running an x86_64 environment, this is likely not the issue.

thundron, 2019-04-09:
@matrixes We made some changes to the way we build packages as well as updating `electron-builder` version, could you please try with the version now available here in the releases section?

closed 2021-07-08 by matrixes