balena-io/etcher #2643: Blurred background in the DMG (macOS installer)

zrzka, 2019-02-22:
See attached image. Repo contains `background.png` and `background@2x.png`, but the DMG contains just TIFF (`background.tiff`). And it leads to a blurred background image. Compare _balenaEtcher.app_ & _Applications_ with _An open source project by balena_.

[image]

lurch, 2019-02-22:
I've no idea what `tiffutil` does, and I don't have a Mac myself, but the `Makefile` _seems_ to be implying that `background.tiff` is made from both `background.png` and `background@2x.png` ?
https://github.com/balena-io/etcher/blob/master/Makefile#L108

zrzka, 2019-02-22:
`tiffutil` in the `Makefile` is supposed to build one `.tiff` image from both `background.png` (standard) and `background@2x.png` (hidpi) images. But the repository already contains `background.tiff` with the same date & time and thus `make assets/dmg/background.tiff` does nothing:

```sh
(master {1} ✓) etcher make assets/dmg/background.tiff 
make: `assets/dmg/background.tiff' is up to date.
(master {1} ✓) etcher
```

And the `background.tiff` in the repo doesn't contain hidpi version, just the standard one.

[image]

To update it, one has to touch either file:

```
(master {1} ✓) etcher touch assets/dmg/background.png 
(master {1} ✓) etcher make assets/dmg/background.tiff 
tiffutil -cathidpicheck assets/dmg/background.png assets/dmg/background@2x.png -out assets/dmg/background.tiff
2 images written to assets/dmg/background.tiff.
(master {1} U:1 ✗) etcher
```

And now, the `background.tiff` contains both versions.

[image]

lurch, 2019-03-11:
I guess this can be closed, it seems it was included in #2674 ?

closed 2019-04-03 by zvin