balena-io/etcher #2644: Unclear why the password is required (macOS)

zrzka, 2019-02-22:
See attached screenshot. Whenever I try to flash an image, dialog appears asking for my password. It says _balenaEtcher wants to make changes._ Good practice on macOS is to tell the user why the password is required. This message is pretty unclear to me.

[image]

schlomo, 2019-09-24:
Bump?

Better would be to also allow granting admin access with Touch ID.

Better yet would be an option to permanently grant admin access so that people who write dozens or hundreds of images every day don't wear out their keyboard typing in the admin password.

lurch, 2019-09-24:
> Better would be to also allow granting admin access with Touch ID.

See #1579 

> Better yet would be an option to permanently grant admin access

IIRC that was asked about before (on Windows I think), and it was rejected because it goes against OSes "security recommendations".

Tech-Nomad, 2020-04-26:
I'm also very suspicious about this prompt. There is no obvious reason why writing to a non-encrypted flash drive would require root access to my whole f*cking machine. And the fact they didn't bother to address this issue just proofs untrustworthy. Why the f*ck does Ubuntu advertise such a tool on their site?

lurch, 2020-04-27:
> There is no obvious reason why writing to a non-encrypted flash drive would require root access

Because Etcher needs to write to the flash drive at the "block level", which is _below_ the "filesystem level". Accessing a drive at the raw block level **always** needs root access, because otherwise it would bypass any filesystem-level access permissions.

schlomo, 2020-04-27:
While I understand this reason I would really prefer a solution where I can once grant Etcher the permission to write to removable media (for example) so that I don't have to type my password for every image.

If this was Linux then I would say: Install an udev rule that grants block-level write permission to the user :-) Don't know so much tech details about OSX, unfortunately.

Tech-Nomad, 2020-04-27:
@schlomo I don't think it is the right thread for your issue. This issue is exactly about the opposite of yours... 

@lurch 
Yes, indeed. I found it out myself yesterday. The USB-drive I wanted to write to is in the /dev directory and has root:staff as owners. So it seems to be more a flaw on the MacOs side. They should provide all the necessary information why and what for the root password is required. 

But because I don't have the information about the purpose of the prompt provided by MacOs. I don't want to give any random app my password. So I've just used the dd command installed natively on the Mac. 

https://askubuntu.com/a/377561
https://askubuntu.com/a/579615

Another option might be to change the ownership of the USB drive??? Didn't try it though.

schlomo, 2020-04-27:
@Tech-Nomad I know that this is thread is for macOS and I want to have a solution for macOS where I can grant Etcher the permission to write to removable media.

On Linux I know how to do this via udev, on macOS I don't know how to achieve this. Maybe I wasn't clear enough before, hope this clarifies it.

Tech-Nomad, 2020-04-27:
Yeah, but the OP wants to know why the password is required while you obviously don't care. In opposite you want to grand the app an eternal root permission. So again: your issue is about something totally different and thus totally off topic.

lurch, 2020-04-27:
@schlomo I don't have a Mac myself, but IIRC on Linux if you run Etcher with `sudo` then the app "already has" root access, so it doesn't need to prompt for the password when writing the image? Maybe the same thing will work on macOS?

miqmago, 2020-06-30:
In my case I've seen this image and my first thought was: which password should I enter?
Seems suspicious to be asked for the computer admin password in an internal application dialog of an open source application just downloaded!! First thing I did is to download again to see if I downloaded an infected branch or something similar. Then I ended up on this issue so I've entered the password but still worried and scared...

Will see if I end up with a completely erased hdd and someone asking bitcoins for the information.

[image]

thundron, 2020-06-30:
We're working on a PR with a slightly better message in the dialogue so that it's clearer what we need the password for (access the drive to flash it), but that's as far as we can get for now

guidolx, 2020-10-02:
I installed Etcher today and was also worried about the application asking my password and right after I entered my password the native dialog pops up asking again the password. You wonder if you didn't give away the password to some hacker.

lurch, 2020-10-02:
@guidolx see #3309 and related issues

thundron, 2020-10-02:
We now have a better dialogue text that is clear enough on the intentions of the application, which is as far as we can go now as mentioned before.
Note that this is necessary only for macOS Catalina and above.

Closing

closed 2020-10-02 by thundron

jamiebriant, 2021-06-24:
This is not the appropriate way to elevate privs in macos. No application should ever ask for a password. The application should request elevation from the OS, and OS will ask me for my password. The only reason to ask me for my password is because you want to exfiltrate my password.

lurch, 2021-06-24:
> The only reason to ask me for my password is because you want to exfiltrate my password.

If you _really_ believe that, then you can try checking the Etcher source-code to see exactly where the "exfiltration" is being done... :wink:

schlomo, 2021-06-24:
While I didn't suspect Etcher of exfiltrating my password I also would prefer Etcher to use a better way to get the necessary permissions, both for security and for convenience.

Other tools manage this so I must assume that it is possible. The main benefit for users is not only that they don't need to give their password to Etcher but - much more important IMHO - that they can use the fingerprint sensor to authorize the access.

Another benefit is that Etcher might be able to keep the permissions so that the users doesn't have to approve it each and every time.

As I am no MacOS developer I unfortunately cannot provide a patch for this.

jamiebriant, 2021-06-25:
> > The only reason to ask me for my password is because you want to exfiltrate my password.
> 
> If you _really_ believe that, then you can try checking the Etcher source-code to see exactly where the "exfiltration" is being done... 😉

1. That's quite hard to actually verify isn't it, and yes, I do this for a living.
2. Even if I verified the source code, how would I know that the binary I downloaded was compiled from that source?
3. "Oh you can build it yourself" - see 1.

lurch, 2021-06-25:
4. If you don't trust Etcher, you can always go back to using `dd` ? :shrug: 

I'm gonna stop replying now, as a) I'm not an Etcher developer (or Balena employee) b) I'm not a MacOS developer.

thundron, 2021-07-07:
Reasons for us to require the password have been specified enough and well enough. Feel free to check the code 👍 locking the issue as it's not constructive anymore