balena-io/etcher #2645: Ask for the password once (macOS)

zrzka, 2019-02-22:
Not sure if this is possible with the current architecture (didn't check the code). But it would be nice if we can ask for the password once. It asks for my password every single time I hit the _Flash!_ button. It's not very friendly when I'd like to flash multiple cards. Some helper tool with elevated permissions (ask once) would be nice.

[image]

lurch, 2019-02-22:
IIRC when we discussed this a couple of years ago, it was considered better security practice to only keep the "password enabled" part running for as short a time as possible, and to forget the password afterwards? (to reduce the chance of possible exploits)
But I might be misremembering or confusing it with something else... :confused: 

On a related note, see #1819

shawaj, 2020-12-14:
As of v1.5.112 on macOS Big Sur I was able to flash multiple SD cards without entering password again.

I'm not sure if there is a specific timeout set on this though.

@thundron @zvin are you able to provide any more info on how this works?

zvin, 2020-12-14:
It uses `sudo` @shawaj

shawaj, 2020-12-14:
@zvin what I meant was, is there an amount of time for which it will not ask you again for your password after the first time?

grafst, 2023-10-18:
It would be nice to allow it just once.