balena-io/etcher #2647: White screen appears when I remove SD card (macOS)

zrzka, 2019-02-22:
* Select an image, drive and flash the SD card
* Wait till _Flash Complete!_ appears (and do nothing in the app, just keep it as it is)
* Remove flashed SD card from the reader
* And white screen appears

Is this reproducible? Trying, but no, it isn't.

[image]

- **Etcher version: 1.5.0** 
- **Operating system and architecture: macOS 10.14.3 & x86_64** 
- **Image flashed: RPi 3B+ 2.29.2+rev1** 
- **Do you see any meaningful error information in the DevTools?** Didn't check :/

zrzka, 2019-02-25:
Another one, after 30-40x tries. It seems to me that something crashed hard.

[image]

lurch, 2019-02-26:
Looks like it's probably related to this: https://github.com/balena-io-modules/drivelist/issues/329

I guess it must be hard to reproduce in Etcher because it's tricky to pull out the SD card at the **exact** same time that `drivelist` is running? :wink:

mifi, 2019-02-26:
I had the same problem with the electron application I'm writing, but it is very important with a stable application even though devices are refreshed all the time, so I ended up wrapping the call to drivelist.list() in a `child_process.fork`, so that if the call crashes, it only crashes a different node.js process, and the crash can be neatly caught in JS-land and retried.

```
const forkWorker = require('fork-worker');
const isDev = require('electron-is-dev');

async function listDrives() {
  const path = `${isDev ? '.' : process.resourcesPath}/src/drives-worker.js`;
  const { drives } = await forkWorker(path);
  return drives;
}
```

(I also wrote the module `fork-worker` for this purpose)

Anyhow 😅 the best solution is of course to fix it upstream in `drivelist`, by someone who knows c++ 🙈

zrzka, 2019-02-26:
@mifi just to be sure, macOS as well?

I'll check the C++ code.

mifi, 2019-02-26:
Yes macos. I filed the issue upstream

zrzka, 2019-02-27:
Should be fixed once the [drivelist PR](https://github.com/balena-io-modules/drivelist/pull/330) is reviewed & merged and etcher dependencies are updated.

closed 2019-05-09 by zvin