balena-io/etcher #2648: It looks like this is not a bootable image warnings disappears when I move the main window (macOS)

zrzka, 2019-02-22:
* Create a bad image (`touch empty.zip`)
* Select it
* Attention modal dialog appears (It looks like this is not a bootable image)
* Grab the main etcher window and move it
* Attention modal dialog disappears

https://www.dropbox.com/s/qgtt7ztfpusyplx/warning-disappears.mov?dl=0

- **Etcher version: 1.5.0** 
- **Operating system and architecture: macOS 10.14.3**

zrzka, 2019-02-22:
It seems that it applies to any modal. As I just tried the same thing with the Settings -> Unsafe mode modal. Same thing did happen.

lurch, 2019-02-24:
Does it actually "disappear", or does it just get moved to behind the main Etcher window?
I've just tested, and the latter happens under Linux (with balenaEtcher 1.4.9), but I agree it would probably make sense for the error modal to remain visible rather than getting hidden behind the main window. (and while the error modal is still open behind the main window, none of the buttons in the main window are functional)

zrzka, 2019-02-24:
For me (macOS), it disappears. Every other control (buttons) in the main window work when it disappears.

closed 2019-04-04 by thundron