balena-io/etcher #2657: Can you update the Etcher debian repository with version 1.5.0? Please and Thank You.

dilbert-tctc, 2019-02-24:
Can you update the Etcher debian repository with version 1.5.0? Please and Thank You.

_Originally posted by @dilbert-tctc in https://github.com/balena-io/etcher/issues/2540#issuecomment-466789421_

lurch, 2019-02-25:
See https://github.com/balena-io/etcher/pull/2635#issuecomment-466913537

theVKC, 2019-03-22:
Hello! It's possible I have a similar issue running Ubuntu MATE 18.04 and attempting to add Etcher using Software Boutique.

When I select Etcher, I notice it appears to be the old Etcher referenced, including no mention of Balena.io.

[image]

When I attempt to update my sources in Fixes>Update Sources List, I am receiving a GPG error:
[image]

Lastly, if I attempt to install the older package anyway, I receive a package-not-available error:
[image]

As I said, I'm not totally sure that this is related to the Debian repo issue per se, but I assumed that this would be a good place to start with a comment and see where we should go. If you'd prefer I open up a new issue, let me know.

Thank you! I :heart: Etcher and I appreciate all that you are doing with the project!

lurch, 2019-03-23:
As part of the switch from resin to balena, I think the repo URL and GPG keys changed also (I tripped over this myself). I think the fix is to delete any existing resin or etcher related repos from your software sources, and re-follow the steps in https://github.com/balena-io/etcher#debian-and-ubuntu-based-package-repository-gnulinux-x86x64

And WRT to the original issue, AFAICT balena-etcher-electron 1.4.9 is still the latest version published to the DEB repository.

dilbert-tctc, 2019-03-31:
Now that you have released a stable version of Etcher 1.5.19 can you please add it to the deb repo? Some of us Ubuntu, Debian and Fedora users prefer to use repos. Thanks.

thundron, 2019-04-08:
Repos (deb | rpm) have been updated

closed 2019-04-08 by thundron