balena-io/etcher #2659: etcher can't no display

goooglethink, 2019-02-25:
1 years
i download this software
every people can use it 
but i just.........
[image]

what is hapen ,........................... :(

bartisimos, 2019-03-03:
some here - no display - no response from Development tool. 

both blank. 
Win 10 
i5 - NVidea - 8 gb
perhaps you have an idea.

lurch, 2019-03-03:
Similar issues... in case it helps the devs track down a common problem
https://github.com/balena-io/etcher/issues/2456
https://github.com/balena-io/etcher/issues/1752
https://github.com/balena-io/etcher/issues/1031
https://github.com/balena-io/etcher/issues/2569
https://github.com/balena-io/etcher/issues/2397
https://github.com/balena-io/etcher/issues/2596
https://github.com/balena-io/etcher/issues/129

bartisimos, 2019-03-04:
short feedback after testing more. Virus Scanner stopped the application - all centent. 
After disabling the virus scanner - evola - screen appeared.
Funny enough - downloading the file triggers as well now the antivirus - potential trojan infection.

lurch, 2019-03-04:
Yeah, sometimes the virus scanners are a bit over-keen on flagging up Etcher, but I think it's simply because it's an executable they've not seen before #2392 #2289 #2291 #862 etc.

edricus, 2019-04-29:
I'm on archlinux same problem and I can't disable my "virus scanner" I don't have it

crackingthecodethomas, 2019-08-02:
Have exact same issue. Win 10 Pro. Tried clean install. Still white blank window.

mac1494, 2019-12-15:
same issue windows 7 64bit clean installed.

koon95, 2020-04-08:
run it on compatibility mode. that should fixed it

rossssco, 2020-05-21:
uname -a 

```
Linux i5-dell 5.3.0-53-generic #47~18.04.1-Ubuntu SMP Thu May 7 13:10:50 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```
getting the same error on version 
balenaEtcher-1.5.90-x64.AppImage
(i.e diplays for a second then blank)

Version 
balenaEtcher-1.5.89-x64.AppImage
works fine.

Iv attached full logs from the dev console
[210520_etcher_error_v1.5.90.zip](https://github.com/balena-io/etcher/files/4661912/210520_etcher_error_v1.5.90.zip)

lurch, 2020-05-21:
@rossssco see #3160 (I believe the Etcher devs are already fixing this)

thundron, 2020-05-21:
> run it on compatibility mode. that should fixed it

I think this is the solution for the o.g. issue
Closing, and @rossssco a new release is in the way to fix yours

closed 2020-05-21 by thundron