balena-io/etcher #2661: Hide mounted DMGs from available drives

zrzka, 2019-02-25:
Would be nice if we can hide mounted DMGs on macOS.

[image]

It doesn't make much sense to have them listed there. This is mounted balenaEtcher 1.5.0 DMG. Which is reported by `drivelist` as:

```js
{ enumerator: 'DiskArbitration',
    busType: 'Virtual Interface',
    busVersion: null,
    device: '/dev/disk4',
    devicePath: 'IODeviceTree:/',
    raw: '/dev/rdisk4',
    description: 'Apple UDIF read-only compressed (zlib) Media',
    error: null,
    size: 311051776,
    blockSize: 512,
    logicalBlockSize: 512,
    mountpoints:
     [ { path: '/Volumes/balenaEtcher 1.5.0',
         label: 'balenaEtcher 1.5.0' } ],
    isReadOnly: true,
    isSystem: false,
    isVirtual: true,
    isRemovable: true,
    isCard: false,
    isSCSI: false,
    isUSB: false,
    isUAS: null }
```

- **Etcher version: 1.5.0** 
- **Operating system and architecture: macOS 10.14.3 & x86_64**

lurch, 2019-02-25:
Looks like a regression in drivelist? https://github.com/balena-io-modules/drivelist/pull/55

See also https://github.com/balena-io/etcher/issues/2597#issuecomment-458248422

zrzka, 2019-02-26:
For the reference, couple of disks on my Mac ...

1. Internal SSD
2. Volume on internal SSD
3. External USB drive
4. SD card
5. Mounted DMG

| Property | 1 (SSD) | 2 (SSD) | 3 (USB) | 4 (SD) | 5 (DMG) |
| -- | -- | -- | -- | -- | -- |
| busType | `PCI-Express` | `PCI-Express` | `USB` | `USB` | `Virtual Interface`
| isReadonly | `false` | `false` | `false` | `false` | `true` |
| isSystem | `true` | `true` | `false` | `false` | `false` |
| isVirtual | `true` | `true` | `true` | `true` | `true` |
| isRemovable | `false` | `false` | `false` | `true` | `true` |
| isCard | `false` | `false` | `false` | `false` | `false` |
| isSCSI | `false` | `false` | `false` | `false` | `false` |
| isUSB | `false` | `false` | `true` | `true` | `false` |
| isUAS | `null` | `null` | `null` | `null` | `null` |

Based on this, we can say that DMG image can be identified as:

* `isReadonly && isRemovable && !isUSB`
* I'd probably add `&& busType == 'Virtual Interface'` (but it can break in the future)

**Internal SSD**

```js
{ enumerator: 'DiskArbitration',
    busType: 'PCI-Express',
    busVersion: null,
    device: '/dev/disk0',
    devicePath: 'IODeviceTree:/PCI0@0/RP17@1B/SSD0@0/IONVMeController',
    raw: '/dev/rdisk0',
    description: 'APPLE SSD SM0256L Media',
    error: null,
    size: 251000193024,
    blockSize: 4096,
    logicalBlockSize: 4096,
    mountpoints: [],
    isReadOnly: false,
    isSystem: true,
    isVirtual: true,
    isRemovable: false,
    isCard: false,
    isSCSI: false,
    isUSB: false,
    isUAS: null }
```

**Volume on internal SSD**

```
{ enumerator: 'DiskArbitration',
    busType: 'PCI-Express',
    busVersion: null,
    device: '/dev/disk1',
    devicePath: 'IODeviceTree:/PCI0@0/RP17@1B/SSD0@0/IONVMeController',
    raw: '/dev/rdisk1',
    description: 'AppleAPFSMedia',
    error: null,
    size: 250685575168,
    blockSize: 4096,
    logicalBlockSize: 4096,
    mountpoints:
     [ { path: '/', label: 'Macintosh HD' },
       { path: '/private/var/vm', label: 'VM' } ],
    isReadOnly: false,
    isSystem: true,
    isVirtual: true,
    isRemovable: false,
    isCard: false,
    isSCSI: false,
    isUSB: false,
    isUAS: null }
```

**External USB drive**

```js
{ enumerator: 'DiskArbitration',
    busType: 'USB',
    busVersion: null,
    device: '/dev/disk2',
    devicePath:
     'IODeviceTree:/PCI0@0/PEG2@1,2/UPSB@0/DSB1@1/UPS0@0/pci-bridge@1/pci1b73,1100@0',
    raw: '/dev/rdisk2',
    description: 'Samsung Portable SSD T3 Media',
    error: null,
    size: 250059350016,
    blockSize: 512,
    logicalBlockSize: 512,
    mountpoints: [ { path: '/Volumes/Zrzka', label: 'Zrzka' } ],
    isReadOnly: false,
    isSystem: false,
    isVirtual: true,
    isRemovable: false,
    isCard: false,
    isSCSI: false,
    isUSB: true,
    isUAS: null }
```

**SD card**

```js
{ enumerator: 'DiskArbitration',
    busType: 'USB',
    busVersion: null,
    device: '/dev/disk3',
    devicePath: 'IODeviceTree:/PCI0@0/PEG2@1,2/UPSB@0/DSB2@2/XHC3@0',
    raw: '/dev/rdisk3',
    description: 'SANDISK SDDR-409 Media',
    error: null,
    size: 31914983424,
    blockSize: 512,
    logicalBlockSize: 512,
    mountpoints: [ { path: '/Volumes/resin-boot', label: 'resin-boot' } ],
    isReadOnly: false,
    isSystem: false,
    isVirtual: true,
    isRemovable: true,
    isCard: false,
    isSCSI: false,
    isUSB: true,
    isUAS: null }
```

**Mounted DMG**

```js
{ enumerator: 'DiskArbitration',
    busType: 'Virtual Interface',
    busVersion: null,
    device: '/dev/disk4',
    devicePath: 'IODeviceTree:/',
    raw: '/dev/rdisk4',
    description: 'Apple UDIF read-only compressed (zlib) Media',
    error: null,
    size: 311051776,
    blockSize: 512,
    logicalBlockSize: 512,
    mountpoints:
     [ { path: '/Volumes/balenaEtcher 1.5.0',
         label: 'balenaEtcher 1.5.0' } ],
    isReadOnly: true,
    isSystem: false,
    isVirtual: true,
    isRemovable: true,
    isCard: false,
    isSCSI: false,
    isUSB: false,
    isUAS: null }
```

lurch, 2019-02-26:
Would / could `isReadonly && isRemovable && !isUSB` also be true for an SD card with it's write-lock switch enabled and in an internal SD-reader slot instead of in a USB SD-reader? (some laptops have their internal SD slot connected via PCI (would appear as `/dev/mmcblkX` under Linux) and some laptops have their internal SD slot connected via USB (would appear as `/dev/sdX` under Linux))

Seems a bit weird that `isVirtual` is `true` for everything - I wonder what that was _supposed_ to indicate? :man_shrugging:

zrzka, 2019-02-26:
> Would / could `isReadonly && isRemovable && !isUSB` also be true for an SD card with its write-lock switch enabled

With external reader, no (just tried it). With internal reader, no idea, doesn't have a Mac with internal reader.

> Seems a bit weird that `isVirtual` is `true` for everything - I wonder what that was _supposed_ to indicate?

On macOS, it's defined as:

```
device.isVirtual = ![mediaType isEqualToString:@"physical"];
```

Where `mediaType` is a value associated with `kDADiskDescriptionMediaTypeKey`. Perhaps we can use another [keys](https://developer.apple.com/documentation/diskarbitration/diskarbitration_constants) to detect proper device type.

Also it seems that it doesn't work, like `isCard`, which is `false` in case of my SD card, because it's detected via icon set file name:

```objc
  bool IsCard(CFDictionaryRef diskDescription) {
    CFDictionaryRef mediaIconDict = (CFDictionaryRef)CFDictionaryGetValue(
      diskDescription,
      kDADiskDescriptionMediaIconKey
    );
    NSString *iconFileName = (NSString*)CFDictionaryGetValue(
      mediaIconDict,
      CFStringCreateWithCString(NULL, "IOBundleResourceFile", kCFStringEncodingUTF8)
    );

    return [iconFileName isEqualToString:@"SD.icns"];
  }
```

Seems that this ObjC code is outdated as well.

> P.S. Perhaps it makes sense to move this issue to the `drivelist` repo?

From the user point of view - it's an balenaEtcher issue. From the technical point of view, it depends. Is `drivelist` supposed to return writable targets / devices for balenaEtcher? If yes, then it's `drivelist` issue. If no, `drivelist` should return let's say everything and balenaEtcher should filter it, it should stay here.

lurch, 2019-02-26:
> Where `mediaType` is a value associated with `kDADiskDescriptionMediaTypeKey`. Perhaps we can use another [keys](https://developer.apple.com/documentation/diskarbitration/diskarbitration_constants) to detect proper device type.

Hmmm, nice in-depth documentation from Apple there!
https://developer.apple.com/documentation/diskarbitration/kdadiskdescriptionmediatypekey?language=objc
:angry: 

> Seems that this ObjC code is outdated as well.

I suspect it probably worked when it was [first written](https://github.com/balena-io-modules/drivelist/pull/296), but perhaps things have changed in different macOS versions? Also, I dunno if there's a reliable way for a USB MassStorageDevice to indicate to the OS whether it's a SD reader or a USB Flash stick or a USB HDD or a USB SSD, etc. - so perhaps for some USB (or internal) SD readers it _does_ use "SD.icns" but for other USB SD readers it just uses a generic disk icon? :man_shrugging:
(I'm just guessing regarding macOS, but certainly on my Ubuntu PC different SD readers appear with different icons - some of them even appear with a CompactFlash icon!)

zvin, 2019-02-28:
Maybe we should just ignore all `isVirtual: true` devices?

zrzka, 2019-03-01:
Yeah, that's the problem. Drivelist says that everything is virtual. I'll dump the diskutil output (not drivelist) when I'll be at home (weekend, to include TimeMachine disk as well) and will check if there're any differences and if we can fix `isVirtual` and provide better info for etcher[-sdk] for filtering.

zvin, 2019-03-01:
Looks like `DADeviceProtocol` is `Virtual Interface` for mounted dmgs, and `USB` or `SATA` (and maybe others) for the rest. Is it enough to say that a device `isVirtual` @zrzka @jviotti @lurch ?

zvin, 2019-03-01:
For now we're checking for `DAMediaType` to be `physical` but it seems like it's always `null`.

zvin, 2019-03-01:
We can probably consider `DAMediaName` === `AppleAPFSMedia` as virtual as it is a container contained in another disk.

zrzka, 2019-03-01:
Don't know yet, will write a summary over weekend. Will gather info from `diskutil`, Disk Utility & output from `DiskArbitration.framework`.

zvin, 2019-03-01:
@zrzka for filtering out APFS containers we can also remove all devices which `DAMediaPath` starts with the `DAMediaPath` of another device.

zrzka, 2019-03-03:
It seems (after the first look), that the DMG & Time Machine can be filtered out by:

```
DADeviceModel = "Disk Image";
DADeviceProtocol = "Virtual Interface";
```

Both have these values. Maybe we can replace ...

```
device.isVirtual = ![mediaType isEqualToString:@"physical"];
```

... with ...

```
NSString *deviceProtocol = (NSString*)CFDictionaryGetValue(diskDescription, kDADiskDescriptionDeviceProtocolKey);
NSString *deviceModel = (NSString*)CFDictionaryGetValue(diskDescription, kDADiskDescriptionDeviceModelKey);

device.isVirtual = [deviceProtocol isEqualToString:@"Virtual Interface"] || [deviceModel isEqualToString:@"Disk Image"];
```

... and then use `isVirtual` in `etcher-sdk` & `etcher`.

# Dump

Here's the dump of everything that `DiskArbitration.framework` provides (10.14.3).

## Internal SSD

### diskutil

```
/dev/disk0 (internal):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      GUID_partition_scheme                         251.0 GB   disk0
   1:                        EFI EFI                     314.6 MB   disk0s1
   2:                 Apple_APFS Container disk1         250.7 GB   disk0s2
```

### DADiskCopyDescription

```
2019-03-03 11:23:44.277507+0100 dadump[94065:2868718] bsdName: /dev/disk0
2019-03-03 11:23:44.277970+0100 dadump[94065:2868718] {
    DAAppearanceTime = "572881797.308954";
    DABusName = IONVMeController;
    DABusPath = "IODeviceTree:/PCI0@0/RP17@1B/SSD0@0/IONVMeController";
    DADeviceInternal = 1;
    DADeviceModel = "APPLE SSD SM0256L";
    DADevicePath = "IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/RP17@1B/IOPP/SSD0@0/IONVMeController/IONVMeBlockStorageDevice@1";
    DADeviceProtocol = "PCI-Express";
    DADeviceRevision = CXS6AA0Q;
    DADeviceUnit = 1;
    DADeviceVendor = "";
    DAMediaBSDMajor = 1;
    DAMediaBSDMinor = 0;
    DAMediaBSDName = disk0;
    DAMediaBSDUnit = 0;
    DAMediaBlockSize = 4096;
    DAMediaContent = "GUID_partition_scheme";
    DAMediaEjectable = 0;
    DAMediaIcon =     {
        CFBundleIdentifier = "com.apple.iokit.IOStorageFamily";
        IOBundleResourceFile = "Internal.icns";
    };
    DAMediaKind = IOMedia;
    DAMediaLeaf = 0;
    DAMediaName = "APPLE SSD SM0256L Media";
    DAMediaPath = "IODeviceTree:/PCI0@0/RP17@1B/SSD0@0/IONVMeController/IONVMeBlockStorageDevice@1/@1:0";
[... 292 lines cut ...]

    CFRelease(diskDescription);
    CFRelease(disk);
}

void printAllDisksInfo() {
    DASessionRef session = DASessionCreate(kCFAllocatorDefault);
    if (session == nil) {
        return;
    }

    for (int i = 0; i < 10; i++) {
        printDiskInfo(session, [NSString stringWithFormat:@"/dev/disk%d", i]);
    }

    CFRelease(session);
}

int main(int argc, const char * argv[]) {
    @autoreleasepool {
        printAllDisksInfo();
    }
    return 0;
}
```

zrzka, 2019-03-04:
@zvin @thundron This [drivelist#fix-is-virtual-macos](https://github.com/balena-io-modules/drivelist/compare/fix-is-virtual-macos) branch & following `etcher-sdk` patch ...

```patch
diff --git a/lib/scanner/adapters/block-device.ts b/lib/scanner/adapters/block-device.ts
index 4a6e0d2..8af54df 100644
--- a/lib/scanner/adapters/block-device.ts
+++ b/lib/scanner/adapters/block-device.ts
@@ -114,7 +114,9 @@ export class BlockDeviceAdapter extends Adapter {
                                        // Exclude system drives if needed
                                        (this.includeSystemDrives() || !drive.isSystem) &&
                                        // Exclude drives with no size
-                                       typeof drive.size === 'number'
+                                       typeof drive.size === 'number' &&
+                                       // Exclude virtual drives (DMG, TimeMachine, ... on macOS)
+                                       !drive.isVirtual
                                )
                        ) {
                                continue;
```

... fixes this issue & #2671.

If `isVirtual` is excluded, I have **no idea** how this will influence Linux & Windows environments. Should I PR it in this state or are you going to test Linux & Windows with these changes? Or any other way how to fix those two issues?

zvin, 2019-03-04:
@zrzka please PR in this state.
Why is `DADeviceProtocol = "Virtual Interface"` not enough for `isVirtual` ?

We still need another fix for excluding APFS containers.

We don't want to list any virtual drive in etcher-sdk so it should be ok.

zrzka, 2019-03-04:
@zvin can you pls elaborate on `excluding APFS containers`? What's the problem?

zvin, 2019-03-04:
[image]

The second drive isn't a drive but is contained in the first one.
See the first two drives in my comment above https://github.com/balena-io/etcher/issues/2661#issuecomment-468662837

zrzka, 2019-03-04:
@zvin but that's visible only and only if you have _Settings - Unsafe mode_ enabled, right? Common users are not going to enable _Unsafe mode_ and if they do, then ...

# On my Mac

## Unsafe mode - disabled

[image]

## Unsafe mode - enabled

[image]

zvin, 2019-03-04:
Yes, that can only be seen in unsafe mode, it's not a major issue. Still a bug though.

zrzka, 2019-03-04:
@zvin as this one was auto closed via merged PR, I did report it as a separate issue #2678 with links to specific comments.

closed 2019-03-04 by zrzka