balena-io/etcher #2662: Failed verification due to hash mismatch

robmcc1983, 2019-02-25:
- **1.5 0** 
- **64 bit windows 10 home** 
- **libreelec and raspian** 
- **No** 

I'm having a problem when writing images on the verification process.
It says there is a hash mismatch at the end of the process.

I have tried 3 different memory cards and all of them produce the same error and one is new.

I have tried writing the libreelec image but when that failed I tried raspian as well to check it wasn't the image.

Anyone know what this could be
Thanks

robmcc1983, 2019-02-27:
I will answer my own question just so if others have the same problem as I appear to have solved this.
I Disabled windows search and storage service within the services of windows 10.
After that etcher wrote the images fine on all 3 of the memory cards no problem at all.
I then re enabled the services, windows search and storage service and rebooted my computer.
Tried again after this and it failed with hash checksum error again.
I then installed a fresh install of Windows 10 on my computer and not changing any settings etcher allowed me to write to SD card successfully.
I thought great problem solved so went through installing the rest of my programs I use and after installing them all I went to finally setup the SD card I wanted, only to be prompted with the same hash checksum error again in etcher.
So I went through the process of uninstalling programs in reverse order I installed them trying etcher each time.
When I got to Western digital drive utility (which I use for a western digital external drive) the problem went away.
I then reinstalled the WD drive utility software and the etcher problem came back.
So it appears that having western digital drive utility installed on a system and trying to use etcher causes the checksum error during the verification process of etcher.
Either WD drive utility is installing something etcher doesn't like or its altering some sort of removable storage service that etcher doesn't like.

lurch, 2019-02-28:
It sounds like maybe "windows search and storage service" is altering some data (or metadata) on the card, and that's what's causing verification to fail?

robmcc1983, 2019-02-28:
It's western digital drive utility that causes the error.
Uninstalling WD utility or disabling the WD service from task manager stops the error happening during the verification process of etcher.

Spokehedz, 2019-04-09:
It is absolutely WD tools. I kept getting failed writes about 98% of the time, and I didn't even know I had this installed OR that it would cause 'problems'. All my flash drives are coming up okie dokie now.

closed 2019-04-09 by thundron

Bortex19, 2021-04-16:
I had the same problem and it was solved doing what you explained @robmcc1983 
Thank you very much