balena-io/etcher #2665: Prevent sleep/hybernate on Windows while writing

ertankucukoglu, 2019-02-26:
Hello,

I am using 1.4.9 version on Windows 10 Pro 64Bit. I was flashing latest Raspberry Pi image to my micro sd card plugged in a usb converter.

My computer goes to sleep after 5mins. I left it while write operation was running. When I am back, I find computer sleeping. Woke up the computer and I see than operation interrupted.

I suggest to register Etcher to prevent going into sleep mode while there is an active operation.

Thanks & regards,
Ertan

zrzka, 2019-03-01:
I like it and it should be done for all OSes, not just Windows.

uncannyjish, 2020-12-09:
I back this feature and I think this one should be a top priority feature to be added in the next update