balena-io/etcher #2676: Kubuntu 18.10 write issue

omerakgoz34, 2019-03-03:
- **Etcher version: 1.5.0** 
- **Operating system and architecture: Win10 x64** 
- **Image flashed: kubuntu-18.10-desktop-amd64.iso** 

I flashing kubuntu-18.10-desktop-amd64.iso but etcher always give me hash error when flash finished.

lurch, 2019-03-03:
Duplicate of #2662 ?

rosshiga, 2019-03-05:
Etcher version: 1.5.5
Operating system and architecture: Win10 x64
Image flashed: kubuntu-18.10-desktop-amd64.iso

CRC32: C6864171
MD5: 8EC1A9BAF54F58241FC89CF695001887
SHA-1: AD3EA8077D4C3F38951446B6497D2DD4B44068B2
SHA256: 9E675F47819292DD91F476DDA22588868E9A207851EBD72769C0C79766949EF0 (Verified)

Confirmed error: Source and destination checksums do not match.

zvin, 2019-04-03:
@rosshiga does this happen only with a specific drive / sd card or any?