balena-io/etcher #2679: CLI missing?

jedahan, 2019-03-05:
- **Etcher version:** 1.5.8
- **Operating system and architecture:** macOS
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?** 

Trying to use etcher-cli from the commandline, and it looks like the latest published version on npm is 0.2.0 and etcher-image-write supposedly is deprecated.

Mind publishing a version on @balena/etcher-cli on npm so people don't accidentally download out of date ones?

lurch, 2019-03-06:
https://github.com/balena-io/etcher/blob/master/CHANGELOG.md#v150---2019-02-16 says "Removed etcher-cli" :confused:

thundron, 2019-04-08:
The CLI has been discontinued, we now support the desktop application only

closed 2019-04-08 by thundron