balena-io/etcher #2684: Balena Etcher is not updating via repository on Ubuntu 19.04 Disk

serial02, 2019-03-08:
Balena Etcher is not updating via repository on Ubuntu 19.04 Disk

- **Etcher version:** 
    1.4.9
- **Operating system and architecture:** 
   Linux edson-p6540br 5.0.0-7-generic #8-Ubuntu SMP Mon Mar 4 16:27:25 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
   Ubuntu Disco 19.04 Disco Dingo

- **Image flashed:** 

- **Do you see any meaningful error information in the DevTools?** 

Updates are not being performed via repository and when we do command sudo apt update is ignored at (Ign: 9 https://dl.bintray.com/etcher/debian stable InRelease), since the version that is installed is 1.4.9 and the version available on your site is 1.5.9 here: (https://github.com/balena-io/etcher) and version 1.5.5 here: (https://www.balena.io/etcher/) e Aqui: v1.5.9 - 2019-03-05 ( https://github.com/balena-io/etcher/blob/master/CHANGELOG.md )

Something needs to be fixed since the settings I have via the repository are those available at (https://github.com/balena-io/etcher)?

Thanks in advance for your attention and I look forward to your return.

Thank you

Edson Santos

lurch, 2019-03-08:
Duplicate of #2657 ?

thundron, 2019-04-15:
New version `1.5.19` released on deb/rpm repositories too. Closing

closed 2019-04-15 by thundron