balena-io/etcher #2687: No Windows showing on MacOS Mojave

benwiggy, 2019-03-11:
- **Etcher version: 1.55
- **Operating system and architecture: MacOS 10.14.3.
- **Image flashed: Not yet!
- **Do you see any meaningful error information in the DevTools? No.

Application launches, showing menubar, but no Windows appear. Preferences dialog does not show. Main window does not show. About Etcher dialog does show! DevTools window does open, but contains no info. 

Clicking on "Show All Windows" in the Dock brings up "No available windows".

lurch, 2019-03-11:
Similar to #1866 ? (which is on Windows rather than Mac)

benwiggy, 2019-03-11:
Similar. There are other reports of the same problem on Mac on the forum:
https://forums.balena.io/t/etcher-1-5-problems-on-mac/5431

donkoller, 2019-03-21:
Same problem. I am going to go back to the old Etcher app until this is resolved.

gkravecz, 2019-03-23:
Same problem. below is the contents of the balenaEtcher Helper crash file that is created when launching balenaEtcher.

[balenaEtcher Helper_2019-03-22-230904_slick.crash.txt](https://github.com/balena-io/etcher/files/2998977/balenaEtcher.Helper_2019-03-22-230904_slick.crash.txt)

jacobcxdev, 2019-03-27:
Same problem here

iCach0, 2019-04-14:
same problem

Hardcore-fs, 2019-05-02:
does not work on  high sierra either.....

Metal ?

waipeinan, 2019-06-12:
still not working, June 2019?

thundron, 2019-06-12:
@waipeinan We're still trying to reproduce, if you have some logs to share they'd be welcome so we can compare them with other reports

thundron, 2019-06-12:
The good old "did you try turning off and on"! @waipeinan it's not needed for now, but if you happen to see it again you can try sharing some system logs like the ones shared by @jacobcxdev if you find any

waipeinan, 2019-06-12:
indeed... and will do.
thanks

***
*Simon Peter Tyszko*
*www.theculture.net <http://www.theculture.net>*
*www.phlight.org <http://www.phlight.org>*
*Isotopica episodes on Resonance FM
<https://www.mixcloud.com/Resonance/playlists/isotopica/>*
*Isotopica <http://www.theculture.net/feed>*
*instagram <https://www.instagram.com/tyszko.isotopica/>*

tseven, 2019-07-14:
Etcher Version 1.5.51
macOS Mojave 10.14.5
iMac (27-inch, Late 2013)

App starts up.
Menu bar loads, but no main app window is visible.

Crash logs attached.

[balenaEtcher_crash_logs.zip](https://github.com/balena-io/etcher/files/3390304/balenaEtcher_crash_logs.zip)

zvin, 2019-11-06:
@tseven thanks for the crash logs!
It's a SEGFAULT:
```
Thread 27 Crashed:
0   libsystem_platform.dylib      	0x00007fff7df81712 _platform_strlen + 18
1   libc++.1.dylib                	0x00007fff7afc7299 std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::assign(char const*) + 21
2   .io.balena.etcher.UjOtGD      	0x000000010b589faa Drivelist::CreateDeviceDescriptorFromDiskDescription(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, __CFDictionary const*) + 295
3   .io.balena.etcher.UjOtGD      	0x000000010b58a53e Drivelist::ListStorageDevices() + 710
4   .io.balena.etcher.UjOtGD      	0x000000010b58827c DriveListWorker::Execute() + 26
5   libnode.dylib                 	0x00000001058a6fee 0x105897000 + 65518
6   libnode.dylib                 	0x00000001058a73bf 0x105897000 + 66495
7   libsystem_pthread.dylib       	0x00007fff7df8d2eb _pthread_body + 126
8   libsystem_pthread.dylib       	0x00007fff7df90249 _pthread_start + 66
9   libsystem_pthread.dylib       	0x00007fff7df8c40d thread_start + 13
```

It happens in drivelist somewhere [here](https://github.com/balena-io-modules/drivelist/blob/5e1c6742c35ef313003d0ead712947dd4427f2d5/src/darwin/list.mm#L57-L100).

Can you reproduce it reliably?
Can you list all the disks connected to you computer when this happens?

tseven, 2019-11-06:
@zvin 
Yes, this issue occurs every time the app is loaded. It's not usable for me at the moment.
I've tried again with the latest version: v1.5.59. (Same issue)

Here is what diskutil reports:
```
diskutil list                                                                                0 < 09:26:37
/dev/disk0 (internal):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      GUID_partition_scheme                         1.0 TB     disk0
   1:                        EFI EFI                     314.6 MB   disk0s1
   2:                 Apple_APFS Container disk1         1.0 TB     disk0s2

/dev/disk1 (synthesized):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      APFS Container Scheme -                      +1.0 TB     disk1
                                 Physical Store disk0s2
   1:                APFS Volume macOS                   357.9 GB   disk1s1
   2:                APFS Volume Preboot                 44.7 MB    disk1s2
   3:                APFS Volume Recovery                510.4 MB   disk1s3
   4:                APFS Volume VM                      10.7 GB    disk1s4
   5:                APFS Volume www                     2.6 GB     disk1s5

/dev/disk2 (external, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      GUID_partition_scheme                        *1.0 TB     disk2
   1:                        EFI EFI                     209.7 MB   disk2s1
   2: B6FA30DA-92D2-4A9A-96F1-871EC6486200               2.3 MB     disk2s2
   3: 2E313465-19B9-463F-8126-8A7993773801               7.3 MB     disk2s3
   4: FA709C7E-65B1-4593-BFD5-E71D61DE9B02               999.9 GB   disk2s4
   5:                 Apple_Boot Boot OSX                134.2 MB   disk2s5

/dev/disk3 (external, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      GUID_partition_scheme                        *1.0 TB     disk3
   1:                        EFI EFI                     209.7 MB   disk3s1
   2: B6FA30DA-92D2-4A9A-96F1-871EC6486200               2.3 MB     disk3s2
   3: 2E313465-19B9-463F-8126-8A7993773801               7.3 MB     disk3s3
   4: FA709C7E-65B1-4593-BFD5-E71D61DE9B02               999.9 GB   disk3s4
   5:                 Apple_Boot Boot OSX                134.2 MB   disk3s5

/dev/disk4 (external, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      GUID_partition_scheme                        *1.0 TB     disk4
   1:                        EFI EFI                     209.7 MB   disk4s1
   2: B6FA30DA-92D2-4A9A-96F1-871EC6486200               2.3 MB     disk4s2
   3: 2E313465-19B9-463F-8126-8A7993773801               7.3 MB     disk4s3
   4: FA709C7E-65B1-4593-BFD5-E71D61DE9B02               999.9 GB   disk4s4
   5:                 Apple_Boot Boot OSX                134.2 MB   disk4s5

/dev/disk5 (external, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      GUID_partition_scheme                        *1.0 TB     disk5
   1:                        EFI EFI                     209.7 MB   disk5s1
   2: B6FA30DA-92D2-4A9A-96F1-871EC6486200               2.3 MB     disk5s2
   3: 2E313465-19B9-463F-8126-8A7993773801               7.3 MB     disk5s3
   4: FA709C7E-65B1-4593-BFD5-E71D61DE9B02               999.9 GB   disk5s4
   5:                 Apple_Boot Boot OSX                134.2 MB   disk5s5

/dev/disk6 (virtual):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:                  Apple_HFS Raid4                  +3.0 TB     disk6

/dev/disk7 (external, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      GUID_partition_scheme                        *6.0 TB     disk7
   1:                        EFI EFI                     209.7 MB   disk7s1
   2: B6FA30DA-92D2-4A9A-96F1-871EC6486200               2.3 MB     disk7s2
   3: 2E313465-19B9-463F-8126-8A7993773801               7.3 MB     disk7s3
   4: FA709C7E-65B1-4593-BFD5-E71D61DE9B02               6.0 TB     disk7s4
[... 24 lines cut ...]
                                 Physical Store disk9s3
   1:                APFS Volume Xtra2                   237.3 GB   disk10s1
   2:                APFS Volume Preboot                 22.4 MB    disk10s2
   3:                APFS Volume Recovery                529.0 MB   disk10s3

/dev/disk11 (external, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      GUID_partition_scheme                        *6.0 TB     disk11
   1:                        EFI EFI                     209.7 MB   disk11s1
   2: B6FA30DA-92D2-4A9A-96F1-871EC6486200               2.3 MB     disk11s2
   3: 2E313465-19B9-463F-8126-8A7993773801               7.3 MB     disk11s3
   4: FA709C7E-65B1-4593-BFD5-E71D61DE9B02               6.0 TB     disk11s4
   5:                 Apple_Boot Boot OSX                134.2 MB   disk11s5

/dev/disk12 (virtual):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:                  Apple_HFS 12TB_Raid4             +12.0 TB    disk12

/dev/disk13 (external, virtual):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:                  Apple_HFS TM_iMac                +3.0 TB     disk13
                                 Logical Volume on disk9s4
                                 9EA16AD5-580E-4831-9F1D-467D1BF8E6C9
                                 Unlocked Encrypted
```

zvin, 2019-11-06:
@tseven could you please try with the just released [v1.5.62](https://github.com/balena-io/etcher/releases/download/v1.5.62/balenaEtcher-1.5.62.dmg) ?

tseven, 2019-11-06:
That did the trick!

The app window is now visible. I don't have a USB drive handy, but I assume the rest works fine.

Thanks!

zvin, 2019-11-06:
\o/
Thank you very much for testing @tseven

closed 2019-11-06 by zvin