balena-io/etcher #2693: Suspicious network connections

netge, 2019-03-18:
- **Etcher version:1.55** 
- **Operating system and architecture:Debian Stretch** 
- **Image flashed:none** 
- **Do you see any meaningful error information in the DevTools?** 

Etcher opens a multitude of suspicious network connections to companies that are known to track users and sell their data. I find it hard to imagine that all these connections are needed to report errors to the devs.

etcher 4461  user   94u  IPv4  79762      0t0  TCP host:39034->183.34.211.130.bc.googleusercontent.com:https (ESTABLISHED)
etcher 4461  user   98u  IPv4  81741      0t0  TCP host:50232->ec2-35-172-177-65.compute-1.amazonaws.com:https (ESTABLISHED)
etcher 4461  user   99u  IPv4  82987      0t0  TCP host:59328->ec2-34-226-180-131.compute-1.amazonaws.com:https (ESTABLISHED)
etcher 4461  user  100u  IPv4  79764      0t0  TCP host:39038->183.34.211.130.bc.googleusercontent.com:https (ESTABLISHED)
etcher 4461  user  101u  IPv4  82980      0t0  TCP host:49540->15.42.188.35.bc.googleusercontent.com:https (ESTABLISHED)
etcher 4461  user  102u  IPv4  82988      0t0  TCP host:59330->ec2-34-226-180-131.compute-1.amazonaws.com:https (ESTABLISHED)
etcher 4461  user  103u  IPv4  80645      0t0  TCP host:37176->fra16s18-in-f136.1e100.net:https (ESTABLISHED)
etcher 4461  user  104u  IPv4  81838      0t0  TCP host:48484->fra15s28-in-f14.1e100.net:https (ESTABLISHED)
etcher 4461  user  105u  IPv4  82991      0t0  TCP host:59336->ec2-34-226-180-131.compute-1.amazonaws.com:https (ESTABLISHED)
etcher 4461  user  106u  IPv4  82992      0t0  TCP host:59338->ec2-34-226-180-131.compute-1.amazonaws.com:https (ESTABLISHED)
etcher 4461  user  107u  IPv4  82993      0t0  TCP host:59340->ec2-34-226-180-131.compute-1.amazonaws.com:https (ESTABLISHED)
etcher 4461  user  108u  IPv4  82994      0t0  TCP host:58084->fra15s28-in-f4.1e100.net:https (ESTABLISHED)
etcher 4461  user  109u  IPv4  82002      0t0  TCP host:49496->server-143-204-214-81.fra53.r.cloudfront.net:https (ESTABLISHED)
etcher 4461  user  110u  IPv4  82003      0t0  TCP host:38108->zrh04s05-in-f99.1e100.net:https (ESTABLISHED)
etcher 4461  user  111u  IPv4  83011      0t0  TCP host:59878->xx-fbcdn-shv-01-amt2.fbcdn.net:https (ESTABLISHED)
etcher 4461  user  112u  IPv4  80672      0t0  TCP host:43630->25.25.190.35.bc.googleusercontent.com:https (ESTABLISHED)
etcher 4461  user  113u  IPv4  82015      0t0  TCP host:38530->ec2-34-233-254-76.compute-1.amazonaws.com:https (ESTABLISHED)
etcher 4461  user  114u  IPv4  80673      0t0  TCP host:43632->25.25.190.35.bc.googleusercontent.com:https (ESTABLISHED)

WTF!

lurch, 2019-03-18:
See also #2497

thundron, 2019-04-03:
We're aware of this and made a PR for it; basically, there are two webviews and analytics in those is not deactivated as of latest version(s), but will be as soon as [this](https://github.com/balena-io/etcher/pull/2698) is merged (soon, just have to sync it with the other two content websites for the webviews to pick up the new param)

thundron, 2019-04-11:
Solved with 6a9b7395419e2cf978dc6e02fef4e8265a225c7f
Now all webviews include a parameter that reflects the opt-out analytics check in the settings, and external content pages exclude analytics based on that.

closed 2019-04-11 by thundron