balena-io/etcher #2697: source.getInnerSource(...).then(...).catch(...).finally is not a function

matmis, 2019-03-19:
- Etcher version: 1.5.5-1 
- Operating system and architecture: Antergos, intel core i7 6700HQ
- Image flashed: 2018-11-13-raspbian-stretch-lite.img
- Do you see any meaningful error information in the DevTools?:

> /usr/lib/etcher/node_modules/resin-corvus/src/resin-corvus.js:233 {"stack":"TypeError: source.getInnerSource(...).then(...).catch(...).finally is not a function\n    at module.exports.selectImageByPath (file:///usr/lib/etcher/generated/gui.js:8862:15)\n    at file:///usr/lib/etcher/generated/gui.js:8896:15\n    at tryCatcher (/usr/lib/etcher/node_modules/bluebird/js/release/util.js:16:23)\n    at Promise._settlePromiseFromHandler (/usr/lib/etcher/node_modules/bluebird/js/release/promise.js:512:31)\n    at Promise._settlePromise (/usr/lib/etcher/node_modules/bluebird/js/release/promise.js:569:18)\n    at Promise._settlePromise0 (/usr/lib/etcher/node_modules/bluebird/js/release/promise.js:614:10)\n    at Promise._settlePromises (/usr/lib/etcher/node_modules/bluebird/js/release/promise.js:694:18)\n    at _drainQueueStep (/usr/lib/etcher/node_modules/bluebird/js/release/async.js:138:12)\n    at _drainQueue (/usr/lib/etcher/node_modules/bluebird/js/release/async.js:131:9)\n    at Async._drainQueues (/usr/lib/etcher/node_modules/bluebird/js/release/async.js:147:5)\n    at Async.drainQueues (/usr/lib/etcher/node_modules/bluebird/js/release/async.js:17:14)\n    at <anonymous>","message":"source.getInnerSource(...).then(...).catch(...).finally is not a function"}

[image]

exodious, 2019-03-22:
I have the same issue. Same OS and same version.

zvin, 2019-04-03:
The package was probably using electron 1 which is not supported.
You can try using the AppImage instead.

closed 2019-04-03 by zvin