balena-io/etcher #2700: electron2

medioman, 2019-03-20:
- **Etcher version:** 
- **Operating system and architecture:** 
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?** 

Hi,electron on aur is Flagged out-of-date on 2019-03-10 it is possible to compile etcher with electron2?
Thank
Cheers

etcher 1.5.5-2
archlinux

lurch, 2019-03-20:
https://github.com/balena-io/etcher/blob/master/CHANGELOG.md#v150---2019-02-16 :wink:

medioman, 2019-03-20:
Hi,I don't understand, I'm not a great expert.ThankCheers

  ----Messaggio originale----

 
 Data: 20-mar-2019 14.02
 
 A: "balena-io/etcher"<etcher@noreply.github.com>
 
 Cc: "medioman"<sonbello@alice.it>, "Author"<author@noreply.github.com>
 
 

 

 
https://github.com/balena-io/etcher/blob/master/CHANGELOG.md#v150---2019-02-16
  
   ��

medioman, 2019-03-20:
Se solo un grande coglione.Vergognati cretino fa un altro mestiere idiota.

  ----Messaggio originale----

 
 Data: 20-mar-2019 14.13
 

 
 
 

 

 
Closed #2700.

lurch, 2019-03-20:
I was pointing out that according to the Changelog, in "v1.5.0" Etcher "Upgrade to Electron v3." - sorry for being too concise.

lurch, 2019-03-20:
@thundron Why has **resin-ci** made so many edits to @medioman 's last comment??

medioman, 2019-03-20:
@Lurch thanks for answering, but where do I find electron3 in archlinux?
I have a hard head.
I repeat I am not a great expert
Thank

lurch, 2019-03-20:
I've never used Arch Linux myself, and IIRC the Etcher package for Arch was made by a 3rd party?
(I'm not directly involved with the Etcher project myself)

mati865, 2019-03-21:
`electron2` package in Arch is old `electron 2.0` kept for compatibility, current `electron 3.1` will be replaced by [version 4.1](https://www.archlinux.org/packages/community-staging/x86_64/electron/) when it's deemed stable enough.

zvin, 2019-04-03:
electron2 is not supported

closed 2019-04-03 by zvin

medioman, 2019-04-04:
Thank i remove electorn2

  ----Messaggio originale----

 
 Data: 3-apr-2019 19.21
 

 
 Cc: "medioman"<sonbello@alice.it>, "Mention"<mention@noreply.github.com>
 
 

 

 
Closed #2700.

medioman, 2019-05-02:
thak.ector

  ----Messaggio originale----

 
 Data: 21-mar-2019 19.26