balena-io/etcher #2704: Booting in UEFI mode

mxa, 2019-03-22:
- **Etcher version:** 1.5.5
- **Operating system and architecture:** 
- **Image flashed:** Boot repair disk

I can boot the boot repair disk, but I can't really use it because it expects to be booted in EFI mode, not in Legacy boot. Etcher will format the drive in a way that only allows me to boot in legacy boot.

lurch, 2019-03-22:
AFAIK this is a function of the image that you flash and/or your BIOS settings. Etcher just directly copies your image byte-by-byte, it doesn't change it in any way.

curcuz, 2019-03-25:
@mxa can you also provide more info on what `Boot repair disk` is? any link?

mxa, 2019-03-30:
@curcuz Boot-Repair-Disk is a bootable live-USB Image which lets you repair broken bootsector / GRUB / MBR. It can be found at https://sourceforge.net/p/boot-repair-cd/home/Home/

@lurch Boot-Repair-Disk will fail to be able to execute [when booted in legacy mode](https://askubuntu.com/questions/366826/how-do-i-change-the-pc-boot-from-legacy-to-efi-mode-when-boot-repair-warns-me).
My bios will let me choose which mode it shall boot, but if I burn the live-USB with etcher, the UEFI mode isn't available. It works when I use unetbootin or copy the files manually.

lurch, 2019-03-31:
Maybe it's a hybrid ISO? https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos
:man_shrugging: 

Seems weird that it says both "Boot-Repair-Disk, the 'must-have' rescue CD !" and "(do not burn it on CD/DVD if your PC came with Windows8/10)"  :confused:

thundron, 2019-04-11:
@mxa What @lurch said.
Etcher doesn't do any image (or target) manipulation: it just writes what it reads byte-to-byte into the selected target(s).
Also note that `unetbootin` works differently than Etcher, and we even advise to use that in [some situations](https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#how-can-i-configure-persistent-storage)

closed 2019-04-11 by thundron

sevku, 2019-05-29:
This does not seem to be accurate. Etcher has changed the filesystem to iso9660. I haven't been able to boot the stick on a purely UEFI device.

jchisholm59, 2020-01-16:
Giving up on Balena and going back to Rufus.

curcuz, 2020-01-17:
@sevku what image did you flash? it must contain partitions with that file system. As @thundron said, etcher does not manipulate the target, it just writes byte by byte from source to target. The code is here publicly to back this statement :)

suragch, 2024-12-25:
This worked for me:

1. Format USB as FAT and selecting MBR.
2. Use Belena to write the iso image.
3. Set to boot in UEFI only in the target computer BIOS settings.