balena-io/etcher #2707: Small error on website

lurch, 2019-03-26:
I just noticed that at the bottom of https://www.balena.io/etcher/ it says "Etcher GUI is based on the Electron framework, which only supports macOS 10.9 and newer versions." but https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#running-in-older-macos-versions says "Etcher GUI is based on the Electron framework, which only supports macOS 10.10 (Yosemite) and newer versions.".

Related to https://github.com/balena-io/etcher/commit/b6ad6e0a85a84b081327d5adb49d4b2e3164bd8d

lurch, 2019-11-06:
According to https://github.com/balena-io/etcher/pull/2964#discussion_r343234253 I've got it the wrong way around; the website is correct but the `USER-DOCUMENTATION.md` is wrong.

shawaj, 2020-12-13:
Closing this as it appears to be fixed on both the website and the user documentation now.

closed 2020-12-13 by shawaj

lurch, 2020-12-13:
Since I opened this issue, Etcher has migrated to a newer version of Electron, but good to see everything has been ironed out now :slightly_smiling_face:

shawaj, 2020-12-14:
Just noticed it is actually still wrong in the [FAQ file](https://github.com/balena-io/etcher/blob/master/FAQ.md) - will sort a PR