balena-io/etcher #2710: Fails to install on Debian arm64

LordMethenor, 2019-03-26:
- **any:** 
- **Debian (via Crostini) ARM64:** 
I can't install on my arm64 Samsung Chromebook plus. I need this working asap ( a few hours). 
Upon sudo apt-get update, I get a 
Failed to fetch https://deb.etcher.io/dists/1.5.17/etcher/binary-arm64/Packages  404  Not Found
I have tried replacing the stable field in
echo "deb https://deb.etcher.io stable etcher" | sudo tee /etc/apt/sources.list.d/balena-etcher.list
with 'beta', and '1.5.17.' I checked the changelog, and arm64 support was in fact added in 1.4. Is there anything I can do?

RussianNeuroMancer, 2019-06-20:
@thundron 

> We don't have packages for arm64 as of now

Maybe then keep this issue open as valid feature request? I also wanted to use Echer on arm64 desktop and find it's not available.

lurch, 2019-06-20:
See also #2814

jtrobec, 2020-09-13:
On the off chance that anyone else comes across this issue: I was looking for an arm64 build of etcher as well. It was pretty straight forward to follow the instructions linked by @thundron earlier in this thread and do the build. 

I will note that I needed to have node 10, because there was some library that apparently was failing to transpile? (https://github.com/panva/jose/issues/49). On ubuntu 18.04 on an aarch64, the default version of node was 8...I upgraded using https://github.com/nodesource/distributions then did a make clean and rm -r node_modules. Then `make electron-develop` and `npm start` worked, etcher came up, and I was able to flash my card.