balena-io/etcher #2711: It always flash fat-16 format to my SD card

Mon-ius, 2019-03-26:
This week I have flashed 3 SD card with different os. But etcher always format the SD card with FAT-16  format which result in I cannot read it in MACOS.

If I can select which format to the SD card system in the future version?

lurch, 2019-03-27:
Etcher doesn't "format the card" it just copies whatever is in your image file byte-for-byte onto your SD card. So if your SD card is coming out FAT-16 formatted, then the partition in your image file must have been FAT-16 formatted.

I'm surprised to hear that MACOS can't read FAT-16 though :man_shrugging: Have you read https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md ?

Mon-ius, 2019-03-27:
First I init the SD card in MACOS with `diskutil partitionDisk /dev/disk2 MBR FAT32 UNTITLED 0b` , after that the SD card is FAT32 format , then use etcher to flash the img to SD card. If it works with FAT32, then I can at least read the /boot partition but I cannot. It shows FAT16 format after flash.

nalexander50, 2019-03-30:
I'm actually having the same issue. On that latest version of macOS Mojave using the latest version of Etcher. I used macOS Disk Utility to format the SD Card to FAT32, flashed the image, and then macOS says that it cannot read the SD Card because of the format. If I plug the SD Card into a Windows computer, the format is showing as FAT. The SD Card is also bootable from a Raspberry Pi. This leads me to believe that the SD Card is still a version of FAT but must not be FAT32 which is readable by macOS.

nalexander50, 2019-03-30:
Actually, it seems that it may have something to do with the particular image that I am flashing. I used Disk Utility to format as FAT32 again (SD still readable by macOS) and then used `dd` to flash the SD card. After `dd` finished, macOS is reporting that the SD Card is not readable. For future reference, the image is 64bit Hass.io for rPi3B+.

As this evidently has nothing to do with Etcher, my comments can be safely ignored by the maintainers.

Mon-ius, 2019-03-31:
Hi, man. Such things I met 2 years ago with it in my last pc with Linux.  

From my experience it's actually the `Etcher` 's problem that it can only used to format the images for Raspberry Pi with certain images.

 I have also tried other images for the real pc or just called the AMD64 CPU based computer. As an result, it can even not boot after flash the usb stick using this tool. 

What makes me report it not just because it, but this time I cannot even reach the `boot` folder to create the `ssh` file on it to enable the `sshd` service on MacOS system. 

But before I tried it on MACOS, the SD card which is flashed from my last PC 'Linux' could be identified in my MACOS. Hope you can understand what I said.

In conclusion, there are now two bugs in this software.

- It cannot flash the normal images on usb stick for the pc with other platform like X86_64.
- It cannot flash the FAT32 format space in MACOS.

What I suggest is adding extra options to fit it.

If you are only plan for burn the images for raspberry pi, plz just ignore it!

zvin, 2019-04-03:
Etcher does not create partition tables or partitions, it just copies the raw content of your disk image to your disk.

closed 2019-04-03 by zvin

bleedr, 2019-05-02:
I have the same issue! With Mojave! After flashing the card with the latest stretch image, it cannot be opened, no boot. Never had that issue before!