balena-io/etcher #2712: Downloads page links for Linux version are actually sending Mac installers

swordsmanluke, 2019-03-27:
- **Etcher version:1.5.15** 
- **Operating system and architecture: Ubuntu Linux 18.10 ** 

Repro:
1) Using Linux visit https://www.balena.io/etcher/
2) Select Linux version (issue presents on both x64 and x32 versions)
3) Open .zip file
4) Be amazed! You now have Mac version instead of Linux. :)

lurch, 2019-03-27:
Hmmm, that's not what I'm seeing... when I download the Linux x64 version that https://www.balena.io/etcher/ offers me I get a `balena-etcher-electron-1.5.15-linux-x64.zip` and inside the zip is a `balenaEtcher.1.5.15.AppImage` file.

However there _are_ a couple of small problems which @thundron needs to address:
- The zip file also contains an unnecessary `__MACOSX` meta-data folder (perhaps this is what confused @swordsmanluke ?)
- The AppImage inside the zip file is missing the executable bit, which was the whole reason for offering the AppImage-inside-a-zip download in the first place :wink:  (without the executable bit being set, the AppImage can't be run)

mauricepoirrier, 2019-03-27:
- ** Ubuntu Linux 18.10 **

@lurch I ran the AppImage inside the zip file and it is not working either.

thundron, 2019-03-27:
@lurch Nice catch cc @nazrhom ^

thundron, 2019-03-27:
@swordsmanluke Did you try with the latest `1.5.18`?

lurch, 2019-03-27:
@thundron I just tried downloading the 1.5.18 x64 AppImage, and the file inside the zip is now set as executable :+1: 

Unfortunately (on my Ubuntu 14.04 x64 laptop) it seems the "blank grey window" problem is back :frowning_face: 
DevTools console says:
```
ELECTRON_ASAR.js:166 Uncaught Error: /tmp/.org.chromium.Chromium.vYOatV: invalid ELF header
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:166:20)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:740)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:180:18)
    at Module.load (internal/modules/cjs/loader.js:620)
    at tryModuleLoad (internal/modules/cjs/loader.js:559)
    at Function.Module._load (internal/modules/cjs/loader.js:551)
    at Module.require (internal/modules/cjs/loader.js:658)
    at require (internal/modules/cjs/helpers.js:20)
    at Object.<anonymous> (/tmp/.mount_balenazgZjoH/resources/app.asar/node_modules/xxhash/lib/xxhash.js:4)
    at Object.<anonymous> (/tmp/.mount_balenazgZjoH/resources/app.asar/node_modules/xxhash/lib/xxhash.js:52)
    at Module._compile (internal/modules/cjs/loader.js:711)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:722)
    at Module.load (internal/modules/cjs/loader.js:620)
    at tryModuleLoad (internal/modules/cjs/loader.js:559)
    at Function.Module._load (internal/modules/cjs/loader.js:551)
    at Module.require (internal/modules/cjs/loader.js:658)
```

thundron, 2019-03-27:
@lurch Yep we're looking into that issue, it seems to happen on the `.deb` package too

closed 2019-04-01 by zvin